weliem/blessed-android #76: Failed to write <0100> to descriptor of characteristic...INVALID_ATTRIBUTE_VALUE_LENGTH

ppilip, 2021-02-23:
Hello,

I tried with your example app to connect to my BLE device and transfer/receive data from it by using the Serial-Over-GATT service.

I manage to connect to device, but when trying to subscribe to characteristic by calling

`connectedPeripheral.setNotify(MY_SERVICE_UUID, MY_CHARACTERISTIC_UUID, true)`

in onDescriptorWrite callback I receive (in log)

`failed to write <0100> to descriptor of characteristic <MY_CHARACTERISTIC_UUID> for device: <MY_DEVICE_ADDRESS>', status 'INVALID_ATTRIBUTE_VALUE_LENGTH'
`

I tried the same thing (connect, subscribe, send/receive data) with https://github.com/android/connectivity-samples/tree/main/BluetoothLeGatt sample app, and it worked without error.

Is there something I'm missing here?

Thanks in advance.

weliem, 2021-02-24:
You device is rejecting the descriptor write. Can you share a full log so I can see what is happening?

ppilip, 2021-02-26:
Sure, sorry for a delay

> 2021-02-26 18:55:04.999 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:04.999 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:05.000 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:05.002 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:05.003 D/BluetoothLeScanner: Start Scan with callback
2021-02-26 18:55:05.007 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=8 mScannerId=0
2021-02-26 18:55:05.023 I/BluetoothCentralManager: scan started
2021-02-26 18:55:06.162 I/BluetoothHandler: Found peripheral 'MY_DEVICE'
2021-02-26 18:55:06.162 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:06.163 D/BluetoothLeScanner: Stop Scan with callback
2021-02-26 18:55:06.171 I/BluetoothCentralManager: scan stopped
2021-02-26 18:55:06.281 I/BluetoothPeripheral: connect to 'MY_DEVICE' (34:C9:F0:81:30:24) using TRANSPORT_LE
2021-02-26 18:55:06.287 D/BluetoothAdapter: STATE_ON
2021-02-26 18:55:06.291 D/BluetoothGatt: connect() - device: 34:C9:F0:81:30:24, auto: false
2021-02-26 18:55:06.291 D/BluetoothAdapter: isSecureModeEnabled
2021-02-26 18:55:06.292 D/BluetoothGatt: registerApp()
2021-02-26 18:55:06.293 D/BluetoothGatt: registerApp() - UUID=e31b2e21-a127-4a32-8087-e70a9f3120bd
2021-02-26 18:55:06.303 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2021-02-26 18:55:06.790 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=34:C9:F0:81:30:24
2021-02-26 18:55:06.819 I/BluetoothPeripheral: connected to 'MY_DEVICE' (BONDED) in 0.5s
2021-02-26 18:55:06.834 D/BluetoothPeripheral: discovering services of 'MY_DEVICE' with delay of 0 ms
2021-02-26 18:55:06.835 D/BluetoothGatt: discoverServices() - device: 34:C9:F0:81:30:24
2021-02-26 18:55:06.960 D/BluetoothPeripheral: starting bonding with 'MY_DEVICE' (34:C9:F0:81:30:24)
2021-02-26 18:55:07.020 D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_CONSENT (3)
2021-02-26 18:55:07.118 D/BluetoothGatt: onConnectionUpdated() - Device=34:C9:F0:81:30:24 interval=6 latency=0 timeout=500 status=0
2021-02-26 18:55:07.119 D/BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2021-02-26 18:55:07.299 D/BluetoothGatt: onSearchComplete() = Device=34:C9:F0:81:30:24 Status=0
2021-02-26 18:55:07.300 I/BluetoothPeripheral: discovered 3 services for 'MY_DEVICE'
2021-02-26 18:55:07.301 I/BluetoothHandler: connected to 'MY_DEVICE'
2021-02-26 18:55:07.302 I/BluetoothPeripheral: setting preferred Phy: tx = LE_2M, rx = LE_2M, options = NO_PREFERRED
2021-02-26 18:55:07.303 D/BluetoothGatt: onPhyUpdate() - status=6 address=34:C9:F0:81:30:24 txPhy=2 rxPhy=2
2021-02-26 18:55:07.304 E/BluetoothPeripheral: update Phy failed, status 'REQUEST_NOT_SUPPORTED'
2021-02-26 18:55:07.304 D/BluetoothGatt: requestConnectionPriority() - params: 0
2021-02-26 18:55:07.305 D/BluetoothPeripheral: requesting connection priority BALANCED
2021-02-26 18:55:07.361 D/BluetoothGatt: onConnectionUpdated() - Device=34:C9:F0:81:30:24 interval=9 latency=0 timeout=360 status=0
2021-02-26 18:55:07.361 D/BluetoothPeripheral: connection parameters: interval=11.3ms latency=0 timeout=3s
2021-02-26 18:55:07.453 D/BluetoothGatt: onConnectionUpdated() - Device=34:C9:F0:81:30:24 interval=39 latency=0 timeout=500 status=0
2021-02-26 18:55:07.453 D/BluetoothPeripheral: connection parameters: interval=48.8ms latency=0 timeout=5s
2021-02-26 18:55:08.721 D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_CONSENT (3)
2021-02-26 18:55:10.734 D/BluetoothPeripheral: bonded with 'MY_DEVICE' (34:C9:F0:81:30:24)
2021-02-26 18:55:14.127 D/BluetoothGatt: setCharacteristicNotification() - uuid: 00005501-d102-11e1-9b23-00025b00a5a5 enable: true
2021-02-26 18:55:14.231 E/BluetoothPeripheral: failed to write <0100> to descriptor of characteristic <00005501-d102-11e1-9b23-00025b00a5a5> for device: '34:C9:F0:81:30:24', status 'INVALID_ATTRIBUTE_VALUE_LENGTH' 
2021-02-26 18:55:14.232 E/BluetoothHandler: ERROR: Changing notification state failed for 00005501-d102-11e1-9b23-00025b00a5a5 (INVALID_ATTRIBUTE_VALUE_LENGTH)

weliem, 2021-02-26:
This looks strange. Your peripheral is already bonded when you connect, but you still get 2x a pairing request! The last one coincides with turning on the notify.

So looks like you peripheral has issues with bonding. Did you make your own peripheral?

ppilip, 2021-02-27:
No, I haven't made the peripheral, but could try to contact Firmware developers.

The double bonding happens on Samsung S10 with Android 11. Bonding dialog is also show 2x in Google BluetoothLeGatt example app, where subscribing works.
The double bonding dialog is not show on Samsung S7 with Android 7.0 (the same error is with your lib, and all works with Google example)

I removed code that subscribes immediately from your example in `onServicesDiscovered`, and I'm calling 
`connectedPeripheral.setNotify(MY_SERVICE_UUID, MY_CHARACTERISTIC_UUID, true)`
on a button click few seconds after confirming both bonding.

Anyways, I'll see if any firmware updates can be made regarding bonding, but what's confusing me is that now it works with Google example, and it doesn't with your example.

weliem, 2021-02-28:
The Google app doesnt report the result of the write to the descriptor. Try adding the onDescriptorWrite callback and see what status code is returned....my guess it that it also gives and error.

The value to write is 2 bytes long 0x01 and 0x00. The error suggests your device is expecting more/less bytes than 2. But 2 bytes is correct according to the BLE standard. So it looks like a firmware bug.

weliem, 2021-03-11:
Any update on this?

ppilip, 2021-03-15:
Hi

Sorry for the delay. I've tested it, yes you were right, the same error is with the google example. Device has some issues (if data is sent before the subscribe is set, it will get stuck), so firmware is to be updated.

Anyways, this issue can be closed now.

Thanks again, and sorry for the delay once again.

closed 2021-03-15 by ppilip