weliem/blessed-android #77: Android 11 BLE Connection Disconnected Frequently

Sasikumars41, 2021-02-24:
In Mi devices with android 11 BLE connection not stable. It disconnects frequently. I'm writing characteristics every 10 seconds. I got this error - Connection Time out, Error code 8, Error code 61.

weliem, 2021-02-24:
The connection timeout on Android 11 is 5 seconds in most cases. So if you wait 10 seconds it will indeed disconnect...

Sasikumars41, 2021-02-24:
Within 5 seconds it's connected. But frequently disconnecting.

weliem, 2021-02-24:
A connection timeout means that there has been no BLE communication for some time, usually 5 seconds, and hence the peripheral disconnects. 

So this is 'normal behaviour'....unless your peripheral has been designed to keep the connection open for a long time

Sasikumars41, 2021-02-25:
[image]
Check My log file.

Sasikumars41, 2021-02-25:
Without notification state change, my device got disconnected.

weliem, 2021-02-26:
You are doing very strange things. Why on earth, are you calling setNotify continuously? And the you keep writing the same value....

Normally you only call setNotify once. And then you do nothing for 5 seconds so the connection times out....

Sasikumars41, 2021-02-27:
So how to prevent the connection from closing. I'm using set notify to do some actions to prevent connection closing. But I cannot achieve.

weliem, 2021-02-27:
Normally with BLE you don't keep connections open longer than needed. The normal flow is to connect, do whatever you need to do, and disconnect as soon as possible.

What peripheral are you using and why do you want to keep the connection open? It seems your peripheral isn't designed to keep connections open as it uses a 5 second connection timeout.

Sasikumars41, 2021-02-27:
It's not disconnected on android 10 and below devices. The problem is on android 11. It's a temperature sensor.

weliem, 2021-02-27:
If it is a temperature sensor I would expect it to send a value every second or so. If it doesn't do that you could read the temperature value every second. That way the connection would remain.

Sasikumars41, 2021-02-27:
My devices uses 49535343-8841-43f4-a8d4-ecbe34729bb3 to give data exchanges. There are 4 requests uses the same UUID.

weliem, 2021-02-28:
That is a proprietary characteristic, so you have to consult the documentation of the manufacturer on how to properly use the peripheral to get the temperature.

Sasikumars41, 2021-02-28:
Refer to this link -- https://microchipdeveloper.com/wireless:ble-mchp-transparent-uart-service

weliem, 2021-03-02:
That doesn't help much since it is just a transparent uart service. You still need to know what commands you can give to read the temperature.

Sasikumars41, 2021-03-06:
I'm trying to write a byte to  service(49535343-8841-43f4-a8d4-ecbe34729bb3) within 5 seconds to prevent connection close.  But not able to achieve this.

weliem, 2021-03-10:
I suggest you look into the Android class called Handler. That is what you should use to do something periodically.

Closing this issue since it is not related the Blessed library