weliem/blessed-android #78: onCharacteristicUpdate not getting called for custom Service. setNotify returns true

vinuraj-ign, 2021-02-26:
I am using the same example code. Only changes made are changing the Service ID as below and trying to get the data from the device. Note that I am connecting the device with **scanForPeripheralsWithNames**.

I am able to connect the device. setNotify is success as per the logs. 

`//Custom UUID
    private static final UUID CUSTOM_MEASUREMENT_CHARACTERISTIC_UUID = UUID.fromString("340a1b80-cf4b-11e1-ac36-0002a5d5c51b");
    private static final UUID CUSTOM_SERVICE_UUID = UUID.fromString("02366e80-cf3a-11e1-9ab4-0002a5d5c51b");`

I have tested it with nRF connect app and it is getting the data without any issues.

Shared the logs below for reference. Please let me know if any additional details required. 

`2021-02-26 16:19:10.741 17373-17373/com.welie.blessedexample I/BluetoothPeripheral: connect to 'BlueNRG' (D4:4E:CF:8B:EC:A9) using TRANSPORT_LE
2021-02-26 16:19:10.744 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 432 capacity
2021-02-26 16:19:10.757 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 420 capacity
2021-02-26 16:19:10.757 17373-17373/com.welie.blessedexample E/CheckPermission: _bluetooth code = 10 
2021-02-26 16:19:10.758 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 120 capacity
2021-02-26 16:19:10.759 17373-17373/com.welie.blessedexample D/BluetoothGatt: connect() - device: D4:4E:CF:8B:EC:A9, auto: false
2021-02-26 16:19:10.759 17373-17373/com.welie.blessedexample D/BluetoothGatt: registerApp()
2021-02-26 16:19:10.760 17373-17373/com.welie.blessedexample D/BluetoothGatt: registerApp() - UUID=582c51eb-a5e2-4edb-9003-c51560b297e4
2021-02-26 16:19:10.798 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 180 capacity
2021-02-26 16:19:10.800 17373-17785/com.welie.blessedexample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=13
2021-02-26 16:19:10.861 17373-17785/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 180 capacity
2021-02-26 16:19:12.002 17373-17452/com.welie.blessedexample I/.blessedexampl: ProcessProfilingInfo new_methods=268 is saved saved_to_disk=1 resolve_classes_delay=5000
2021-02-26 16:19:13.752 17373-17785/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=13 device=D4:4E:CF:8B:EC:A9
2021-02-26 16:19:13.774 17373-17785/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 168 capacity
2021-02-26 16:19:13.779 17373-17785/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 168 capacity
2021-02-26 16:19:13.784 17373-17785/com.welie.blessedexample I/BluetoothPeripheral: connected to 'BlueNRG' (NONE) in 3.0s
2021-02-26 16:19:13.788 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 168 capacity
2021-02-26 16:19:13.790 17373-17373/com.welie.blessedexample D/BluetoothPeripheral: discovering services of 'BlueNRG' with delay of 0 ms
2021-02-26 16:19:13.790 17373-17373/com.welie.blessedexample D/BluetoothGatt: discoverServices() - device: D4:4E:CF:8B:EC:A9
2021-02-26 16:19:13.792 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 180 capacity
2021-02-26 16:19:14.351 17373-17785/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=D4:4E:CF:8B:EC:A9 interval=6 latency=0 timeout=500 status=0
2021-02-26 16:19:14.402 17373-17785/com.welie.blessedexample D/BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2021-02-26 16:19:14.445 17373-17785/com.welie.blessedexample D/BluetoothGatt: onSearchComplete() = Device=D4:4E:CF:8B:EC:A9 Status=0
2021-02-26 16:19:14.450 17373-17785/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 168 capacity
2021-02-26 16:19:14.452 17373-17785/com.welie.blessedexample I/BluetoothPeripheral: discovered 3 services for 'BlueNRG'
2021-02-26 16:19:14.456 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 168 capacity
2021-02-26 16:19:14.457 17373-17373/com.welie.blessedexample I/BluetoothHandler: connected to 'BlueNRG'
2021-02-26 16:19:14.462 17373-17373/com.welie.blessedexample D/BluetoothGatt: configureMTU() - device: D4:4E:CF:8B:EC:A9 mtu: 247
2021-02-26 16:19:14.463 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 180 capacity
2021-02-26 16:19:14.465 17373-17373/com.welie.blessedexample I/BluetoothPeripheral: requesting MTU of 247
2021-02-26 16:19:14.474 17373-17785/com.welie.blessedexample D/BluetoothGatt: onConfigureMTU() - Device=D4:4E:CF:8B:EC:A9 mtu=247 status=0
2021-02-26 16:19:14.478 17373-17373/com.welie.blessedexample I/BluetoothHandler: new MTU set: 247
2021-02-26 16:19:14.480 17373-17373/com.welie.blessedexample I/BluetoothPeripheral: setting preferred Phy: tx = LE_2M, rx = LE_2M, options = NO_PREFERRED
2021-02-26 16:19:14.483 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 180 capacity
2021-02-26 16:19:14.484 17373-17785/com.welie.blessedexample D/BluetoothGatt: onPhyUpdate() - status=6 address=D4:4E:CF:8B:EC:A9 txPhy=2 rxPhy=2
2021-02-26 16:19:14.485 17373-17785/com.welie.blessedexample E/BluetoothPeripheral: update Phy failed, status 'REQUEST_NOT_SUPPORTED'
2021-02-26 16:19:14.486 17373-17373/com.welie.blessedexample D/BluetoothGatt: requestConnectionPriority() - params: 1
2021-02-26 16:19:14.490 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 180 capacity
2021-02-26 16:19:14.491 17373-17373/com.welie.blessedexample D/BluetoothPeripheral: requesting connection priority HIGH
2021-02-26 16:19:14.492 17373-17373/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: 340a1b80-cf4b-11e1-ac36-0002a5d5c51b enable: true
2021-02-26 16:19:14.496 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 180 capacity
2021-02-26 16:19:14.498 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 180 capacity
2021-02-26 16:19:14.531 17373-17785/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=D4:4E:CF:8B:EC:A9 interval=39 latency=0 timeout=500 status=0
2021-02-26 16:19:14.533 17373-17785/com.welie.blessedexample D/BluetoothPeripheral: connection parameters: interval=48.8ms latency=0 timeout=5s
2021-02-26 16:19:14.536 17373-17373/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'true' for 340a1b80-cf4b-11e1-ac36-0002a5d5c51b
2021-02-26 16:19:14.949 17373-17373/com.welie.blessedexample D/Parcel: Parcel 0x7fd0b7a090: freeing with 120 capacity
2021-02-26 16:19:14.977 17373-17785/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=D4:4E:CF:8B:EC:A9 interval=12 latency=0 timeout=2000 status=0
2021-02-26 16:19:14.983 17373-17785/com.welie.blessedexample D/BluetoothPeripheral: connection parameters: interval=15.0ms latency=0 timeout=20s`

weliem, 2021-02-26:
I dont see anything strange in the log. If it works with nRF connect I would have a look at their log and see what they do exactly. It could have something to do with the fact that you requested a different MTU...

weliem, 2021-03-10:
Any update on this?

vinuraj-ign, 2021-03-11:
Hi,

Apologies for no response. There were multiple tests going on at the same time and the device was somehow connecting to some another phone. There is no issue with the code and I am able to get the data without any issue. 

Closing.

closed 2021-03-11 by vinuraj-ign