weliem/blessed-android #80: Memory Leak Problem in BluetoothPeripheral

alikhangholi, 2021-03-11:
Hello, there is a little problem with memory leaking in BluetoothPeripheral Class.
there are two BroadCastRecievers registered in this method:

    **private void registerBondingBroadcastReceivers() {
        context.registerReceiver(bondStateReceiver, new IntentFilter(BluetoothDevice.ACTION_BOND_STATE_CHANGED));
        context.registerReceiver(pairingRequestBroadcastReceiver, new IntentFilter(BluetoothDevice.ACTION_PAIRING_REQUEST));
    }**

but there is not any public method to unregister them. so when I wanted to destroy my service wich had had an instance of BluetoothPeripheral class, I saw this error: 

**E/ActivityThread: Service com.barnamechi.cutinoapp.MainForeGroundService has leaked IntentReceiver com.welie.blessed.BluetoothPeripheral that was originally registered here. Are you missing a call to unregisterReceiver()?**

weliem, 2021-03-14:
Hi, the receivers are unregistered when a peripheral disconnects. So try calling cancelConnection()....that should help

weliem, 2021-03-26:
Any update on this?

weliem, 2021-03-30:
Closing this issue...