weliem/blessed-android #84: onCharacteriticsUpdate callback is not running

ArslanKhan99, 2021-04-23:
onCharacteriticsUpdate callback is not running

weliem, 2021-04-23:
Please share a log and explain what you are trying to

ArslanKhan99, 2021-04-26:
BluetoothCentralManagerCallback callback = new BluetoothCentralManagerCallback() {
            @Override
            public void onConnectedPeripheral(@org.jetbrains.annotations.NotNull BluetoothPeripheral peripheral) {
                Toast.makeText(NewActivity.this, "Connected", Toast.LENGTH_SHORT).show();
                
BluetoothGattCharacteristic currentTimeCharacteristic = peripheral.getCharacteristic(UUID.fromString(serviceUUID),UUID.fromString(readNotifyUUID));

                if (currentTimeCharacteristic != null) {
                    peripheral.setNotify(currentTimeCharacteristic, true);
                    BluetoothGattDescriptor descriptor= currentTimeCharacteristic.getDescriptor(UUID.fromString(serviceUUID));
                    descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                    peripheral.writeDescriptor(descriptor,"0100".getBytes());
                }
          
boolean isReading=peripheral.readCharacteristic(UUID.fromString(serviceUUID),UUID.fromString(readNotifyUUID)); (Always false)

                
                if (currentTimeCharacteristic != null){
                byte[]  readValue = currentTimeCharacteristic.getValue();   (Always null)
                   
            }

here is my problem I am not able to read or write anything to my ble lock

weliem, 2021-04-26:
Please have a look at the included example app. You will find everything you need to know there.

You have to realize that every operation is asynchronous. So wait for the callbacks in BluetoothPeripheralCallback.

ArslanKhan99, 2021-04-26:
That is the problem as it do not wait and device disconnected automatically in 2 seconds after connection establish successfully

ArslanKhan99, 2021-04-26:
i used the example first and it disconnected after connection establish

weliem, 2021-04-26:
Then your device probably doesnt have any of the characteristics used in the example.

Can you share a log?

ArslanKhan99, 2021-04-27:
sure I ll share log in a while Kindly note that When I use any other library it give error that characteristics is not readable on read and not writeable on write I am stuck at this from last 20 days

ArslanKhan99, 2021-04-27:
here are logs. Please help me out. I am stuck very badly.

2021-04-27 13:07:27.928 975-1011/? I/ActivityManager: Killing 22436:com.microsoft.skydrive/u0a113 (adj 999): empty #32
        2021-04-27 13:07:28.024 714-714/? I/Zygote: Process 23013 exited due to signal 9 (Killed)
        2021-04-27 13:07:28.025 975-1069/? I/libprocessgroup: Successfully killed process cgroup uid 5022 pid 23013 in 93ms
        2021-04-27 13:07:28.028 762-1659/? I/CameraService: UidPolicy: No processes running for a uid=5022 disabled=0
        2021-04-27 13:07:28.051 19344-23750/com.hait.bleconnectiondemo V/Toast: HIDE: android.widget.Toast$TN@bd00c11
2021-04-27 13:07:28.052 975-975/? I/WindowManager: Reparenting to leash, surface=Surface(name=f64d05c Toast)/@0x890ef65
        2021-04-27 13:07:28.052 519-2139/? I/SurfaceFlinger: id=398 createSurf (0x0),-1 flag=80004, Surface(name=f64d05c Toast)/@0x890ef65 - animation-leash#0
        2021-04-27 13:07:28.053 19344-19344/com.hait.bleconnectiondemo V/Toast: HANDLE HIDE: android.widget.Toast$TN@bd00c11 mView=android.widget.LinearLayout{1688176 V.E...... ........ 0,0-363,116}
        2021-04-27 13:07:28.053 19344-19344/com.hait.bleconnectiondemo V/Toast: REMOVE! android.widget.LinearLayout{1688176 V.E...... ........ 0,0-363,116} in android.widget.Toast$TN@bd00c11
2021-04-27 13:07:28.054 975-975/? D/WindowManager: makeSurface duration=2 leash=Surface(name=Surface(name=f64d05c Toast)/@0x890ef65 - animation-leash)/@0x467a790
        2021-04-27 13:07:28.056 19344-23791/com.hait.bleconnectiondemo D/OpenGLRenderer: destroyEglSurface : 0x7630401100
        2021-04-27 13:07:28.056 19344-23791/com.hait.bleconnectiondemo I/mali_egl: eglDestroySurface() in
        2021-04-27 13:07:28.058 19344-23791/com.hait.bleconnectiondemo I/mali_winsys: delete_surface() [363x116] return
        2021-04-27 13:07:28.058 19344-23791/com.hait.bleconnectiondemo I/mali_egl: eglDestroySurface() out
        2021-04-27 13:07:28.059 519-7376/? E/BufferQueueProducer: [Toast$_19344#1] disconnect: not connected (req=1)
        2021-04-27 13:07:28.059 19344-23791/com.hait.bleconnectiondemo E/OpenGLRenderer: ReliableSurface: perform returned an error
        2021-04-27 13:07:28.059 19344-23791/com.hait.bleconnectiondemo W/libEGL: EGLNativeWindowType 0x769c6c0f90 disconnect failed
        2021-04-27 13:07:28.059 19344-23791/com.hait.bleconnectiondemo D/OpenGLRenderer: ~ReliableSurface : 0x769c6c0f80
        2021-04-27 13:07:28.059 19344-19344/com.hait.bleconnectiondemo I/ViewRootImpl@b98584d[Toast]: dispatchDetachedFromWindow
        2021-04-27 13:07:28.066 975-1010/? D/InputTransport: Input channel destroyed: 'f64d05c', fd=372
        2021-04-27 13:07:28.074 19344-19344/com.hait.bleconnectiondemo D/InputTransport: Input channel destroyed: 'f64d05c', fd=77
        2021-04-27 13:07:28.082 975-1011/? D/ConnectivityService: ConnectivityService NetworkRequestInfo binderDied(NetworkRequest [ LISTEN id=161, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&NOT_VPN&FOREGROUND Uid: 10113] ], android.os.BinderProxy@e906889)
        2021-04-27 13:07:28.083 975-1616/? D/ConnectivityService: ConnectivityService NetworkRequestInfo binderDied(NetworkRequest [ TRACK_DEFAULT id=159, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED Uid: 10113] ], android.os.BinderProxy@64c708e)
        2021-04-27 13:07:28.085 714-714/? I/Zygote: Process 22436 exited due to signal 9 (Killed)
        2021-04-27 13:07:28.086 975-1270/? D/ConnectivityService: releasing NetworkRequest [ LISTEN id=161, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&NOT_VPN&FOREGROUND Uid: 10113] ] (release request)
        2021-04-27 13:07:28.087 975-1270/? D/ConnectivityService: releasing NetworkRequest [ TRACK_DEFAULT id=159, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED Uid: 10113] ] (release request)
        2021-04-27 13:07:28.087 975-1270/? D/ConnectivityService:  Removing from current network NetworkAgentInfo [WIFI () - 601], leaving 44 requests.
        2021-04-27 13:07:28.093 975-1069/? I/libprocessgroup: Successfully killed process cgroup uid 10113 pid 22436 in 61ms
        2021-04-27 13:07:28.099 975-1449/? W/NotificationService: Toast already killed. pkg=com.hait.bleconnectiondemo callback=android.app.ITransientNotification$Stub$Proxy@ac7acbc
2021-04-27 13:07:28.112 762-1659/? I/CameraService: UidPolicy: No processes running for a uid=10113 disabled=0
        2021-04-27 13:07:28.246 24300-24300/? V/ClearcutMetricXmitter: Transmission is done.
        2021-04-27 13:07:28.579 975-1060/? I/WindowManager: Reparenting to original parent: Surface(name=WindowToken{88ba2f6 android.os.Binder@1ecc391})/@0x85daa63, destroyLeash=true, surface=Surface(name=f64d05c Toast)/@0x890ef65
        2021-04-27 13:07:28.594 519-2139/? I/Layer: id=398 removeFromCurrentState Surface(name=f64d05c Toast)/@0x890ef65 - animation-leash#0 (64)
        2021-04-27 13:07:28.600 519-2139/? I/SurfaceFlinger: id=398 Removed Surface(name=f64d05c Toast)/@0x890ef65 - animation-leash#0 (64)
        2021-04-27 13:07:28.602 975-1060/? I/WindowManager: Destroying surface Surface(name=Toast$_19344)/@0xe0f9fc1 called by com.android.server.wm.WindowStateAnimator.destroySurface:1834 com.android.server.wm.WindowStateAnimator.destroySurfaceLocked:733 com.android.server.wm.WindowState.destroySurfaceUnchecked:3642 com.android.server.wm.RootWindowContainer.performSurfacePlacementNoTrace:813 com.android.server.wm.RootWindowContainer.performSurfacePlacement:671 com.android.server.wm.WindowSurfacePlacer.performSurfa [... cut ...]
        2021-04-27 13:07:35.774 24502-24502/? D/ApplicationLoaders: Returning zygote-cached class loader: /system/framework/android.hidl.manager-V1.0-java.jar
        2021-04-27 13:07:35.774 24502-24502/? D/ApplicationLoaders: Returning zygote-cached class loader: /system/framework/android.hidl.base-V1.0-java.jar
        2021-04-27 13:07:35.776 24502-24502/? W/android.dqagen: Insufficient pre-allocated space to mmap vdex.
        2021-04-27 13:07:35.798 24502-24502/? D/DQA: Policy is still valid

ArslanKhan99, 2021-04-28:
kindly response...

ArslanKhan99, 2021-04-29:
Can you tell me how we can allow read and write permission through this library

weliem, 2021-04-29:
I don't see any logging from Blessed. Please add Timber as a dependency and use it in you code.

weliem, 2021-04-29:
Use this to initialize Timber:

        // Plant a tree
        Timber.plant(new Timber.DebugTree());

weliem, 2021-04-29:
After this you log should show info from Blessed.

Please share a log

Sasikumars41, 2021-05-14:
Its due to after "Only wait for completion of requestConnectionPriority on Android 8 or higher due to absence of callback in older versions" in code change.

weliem, 2021-05-14:
I am not sure what you mean. Do you mean it is fixed now?

Sasikumars41, 2021-05-17:
No. The Issue not happened in previous release. But, Current release only have this issue. Check my previous comment for git commit name.

weliem, 2021-05-17:
Which of the 2 calls is causing you problems? RequestMTU shouldn't be a problem. So I think it is the connection priority...

Which phone is giving you issues? Which Android version?

Sasikumars41, 2021-05-17:
Samsung J7 - Android OS 8.0.1, Redmi Note 9 pro - Android OS 11

weliem, 2021-05-17:
Ok thx. I know how to fix it then...

weliem, 2021-05-17:
I did some tests with a Samsung J5 (Android 8.1), a Huawei P20 Lite (Android 8.0.0), a Samsung S20 (Android 11) and a Pixel 5 (Android 11). All work....

So I am a bit puzzled why it is not working on your phones. Can you put a breakpoint on **onConnectionUpdated()** to see if it gets called? Your remark suggests it doesn't get called but that would be strange.

And also a piece of logcat would help...

weliem, 2021-05-17:
On all my phones I see this:

I/BluetoothHandler: connected to 'TAIDOC TD1242'
D/BluetoothGatt: configureMTU() - device: C0:26:DA:13:38:FE mtu: 185
I/BluetoothPeripheral: requesting MTU of 185
D/BluetoothGatt: onConfigureMTU() - Device=C0:26:DA:13:38:FE mtu=23 status=0
I/BluetoothHandler: new MTU set: 23
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothPeripheral: requesting connection priority HIGH
D/BluetoothGatt: onConnectionUpdated() - Device=C0:26:DA:13:38:FE interval=40 latency=0 timeout=500 status=0
D/BluetoothPeripheral: connection parameters: interval=50.0ms latency=0 timeout=5s
D/BluetoothPeripheral: reading characteristic <00002a29-0000-1000-8000-00805f9b34fb>
....

Sasikumars41, 2021-05-18:
Im also getting below log

D/BluetoothGatt: connect() - device: 04:91:62:A5:09:E7, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=750c2dcd-86b1-4607-becf-a01a93ae6846
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=04:91:62:A5:09:E7
D/BluetoothGatt: discoverServices() - device: 04:91:62:A5:09:E7
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=04:91:62:A5:09:E7 Status=0
D/BluetoothGatt: configureMTU() - device: 04:91:62:A5:09:E7 mtu: 185
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=39 latency=0 timeout=500 status=0
D/BluetoothGatt: onConfigureMTU() - Device=04:91:62:A5:09:E7 mtu=158 status=0
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=12 latency=0 timeout=2000 status=0
D/BluetoothGatt: cancelOpen() - device: 04:91:62:A5:09:E7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=04:91:62:A5:09:E7
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=8
D/BluetoothGatt: connect() - device: 04:91:62:A5:09:E7, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=e7bd41b7-77a9-4c77-8acb-aba1da1259fd
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=04:91:62:A5:09:E7
D/BluetoothGatt: discoverServices() - device: 04:91:62:A5:09:E7
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=04:91:62:A5:09:E7 Status=0
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=39 latency=0 timeout=500 status=0
D/BluetoothGatt: configureMTU() - device: 04:91:62:A5:09:E7 mtu: 185
D/BluetoothGatt: onConfigureMTU() - Device=04:91:62:A5:09:E7 mtu=158 status=0
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothGatt: onConnectionUpdated() - Device=04:91:62:A5:09:E7 interval=12 latency=0 timeout=2000 status=0

Below is my Code

// Callback for peripherals
    private final BluetoothPeripheralCallback peripheralCallback = new BluetoothPeripheralCallback() {

        @Override
        public void onServicesDiscovered(@NotNull BluetoothPeripheral peripheral) {
                        // Request a new connection priority
            peripheral.requestConnectionPriority(ConnectionPriority.BALANCED);

            //In order to continuously gets notified about the change in characteristics,
            // we need to set the Notification as ON for the characteristics.
            // In our case, we need to know about the temperature change.
            // So after the discovery callback, here’s how we do it
                    peripheral.setNotify(UUID.fromString(GattAttributes.SERVICE_UUID), UID.fromString(GattAttributes.CHAR_UUID_TX), true);

            readHardwareAddress();
            readDeviceName();
        }

I'm Not able read or write characteristics.

Sasikumars41, 2021-05-18:
After I Commanded the blow line, it gives me the  hardware address

peripheral.requestConnectionPriority(ConnectionPriority.HIGH);

weliem, 2021-05-18:
Ok, I revert the changes in the latest commits. Your code should work now again...

Please confirm

Sasikumars41, 2021-05-19:
Sure.

closed 2021-05-25 by weliem