weliem/blessed-android #85: Test application does not detect HRS device

ma8ra, 2021-04-24:
Hello!

I am trying to run your test application and connect to a peripheral, but the scan does not find a device.

The STM Nucleo board with STM32WB55 on board acts as a peripheral device.

The board is configured as a BLE server and has HRS service 0000180D-0000-1000-8000-00805f9b34fb with the characteristic 00002A37-0000-1000-8000-00805f9b34fb (as in your BluetoothHandler.java file).
The board transmits 10 bpm and this is shown by the ST BLE Sensor test application and another application for heart rate sensors with Play Martet.

The problem is that your test application does not see it (the text "Waiting for measurement ...") and the process is terminated at I / BluetoothCentralManager: scan started

Please tell me what could be the problem?

[image]
[image]

04/24 17:27:25: Launching 'app' on HUAWEI BAH2-L09.
Install successfully finished in 2 s 536 ms.
$ adb shell am start -n "com.example.testble4/com.example.testble4.MainActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 13870 on device 'huawei-bah2_l09-QXDNU20217101864'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/HwApiCacheMangerEx: apicache mCurPackageName=com.example.testble4 uptimes=85509272
D/HwFLClassLoader: get used feature list :/feature/used-list failed!
    USE_FEATURE_LIST had not init! 
I/HwApiCacheMangerEx: apicache oUid null
I/HwApiCacheMangerEx: apicache volumes null
I/HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=com.example.testble4#10122#256
I/HwApiCacheMangerEx: apicache oUid 10122
    apicache volumes null
I/HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=com.example.testble4#10122#0
    apicache async read begin packageName=com.example.testble4 userid=0
I/HwApiCacheMangerEx: apicache pi null
I/HwApiCacheMangerEx: apicache pi null
I/HwApiCacheMangerEx: apicache pi null
I/HwApiCacheMangerEx: apicache pi null
E/Minikin: Could not get cmap table size!
I/HwApiCacheMangerEx: apicache oUid null
I/HwApiCacheMangerEx: apicache async read finished packageName=com.example.testble4 userid=0 totalus=17536
E/MemoryLeakMonitorManager: MemoryLeakMonitor.jar is not exist!
I/HwCust: Constructor found for class android.app.HwCustActivityImpl
D/HwCust: Create obj success use class android.app.HwCustActivityImpl
V/HwPolicyFactory: : success to get AllImpl object and return....
I/HwCust: Constructor found for class android.app.HwCustHwWallpaperManagerImpl
D/HwCust: Create obj success use class android.app.HwCustHwWallpaperManagerImpl
V/HwWidgetFactory: : successes to get AllImpl object and return....
V/ActivityThread: ActivityThread,callActivityOnCreate
D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.23, mControlPoint1y = 0.06, mControlPoint2x = 0.09, mControlPoint2y = 0.97
D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.6, mControlPoint1y = 0.9, mControlPoint2x = 0.8, mControlPoint2y = 1.0
D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.23, mControlPoint1y = 0.06, mControlPoint2x = 0.09, mControlPoint2y = 0.97
D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.6, mControlPoint1y = 0.9, mControlPoint2x = 0.8, mControlPoint2y = 1.0
I/zygote64: Rejecting re-init on previously-failed class java.lang.Class<androidx.core.view.ViewCompat$2>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/view/View$OnUnhandledKeyEventListener;
        at void androidx.core.view.ViewCompat.setBackground(android.view.View, android.graphics.drawable.Drawable) (ViewCompat.java:2678)
        at void androidx.appcompat.widget.ActionBarContainer.<init>(android.content.Context, android.util.AttributeSet) (ActionBarContainer.java:63)
        at java.lang.Object java.lang.reflect.Constructor.newInstance0(java.lang.Object[]) (Constructor.java:-2)
        at java.lang.Object java.lang.reflect.Constructor.newInstance(java.lang.Object[]) (Constructor.java:334)
        at android.view.View android.view.LayoutInflater.createView(java.lang.String, java.lang.String, android.util.AttributeSet) (LayoutInflater.java:658)
I/zygote64:     at android.view.View android.view.LayoutInflater.createViewFromTag(android.view.View, java.lang.String, android.content.Context, android.util.AttributeSet, boolean) (LayoutInflater.java:801)
        at android.view.View android.view.LayoutInflater.createViewFromTag(android.view.View, java.lang.String, android.content.Context, android.util.AttributeSet) (LayoutInflater.java:741)
        at void android.view.LayoutInflater.rInflate(org.xmlpull.v1.XmlPullParser, android.view.View, android.content.Context, android.util.AttributeSet, boolean) (LayoutInflater.java:874)
        at void android.view.LayoutInflater.rInflateChildren(org.xmlpull.v1.XmlPullParser, android.view.View, android.util.AttributeSet, boolean) (LayoutInflater.java:835)
        at android.view.View android.view.LayoutInflater.inflate(org.xmlpull.v1.XmlPullParser, android.view.ViewGroup, boolean) (LayoutInflater.java:515)
        at android.view.View android.view.LayoutInflater.inflate(int, android.view.ViewGroup, boolean) (LayoutInflater.java:423)
        at android.view.View android.view.LayoutInflater.inflate(int, android.view.ViewGroup) (LayoutInflater.java:374)
        at android.view.ViewGroup androidx.appcompat.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:896)
        at void androidx.appcompat.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:806)
        at void androidx.appcompat.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:693)
        at void androidx.appcompat.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:170)
        at void com.example.testble4.MainActivity.onCreate(android.os.Bundle) (MainActivity.java:40)
        at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:7383)
        at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1218)
[... 164 lines cut ...]
        at void android.os.Looper.loop() (Looper.java:166)
        at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:7529)
        at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
        at void com.android.internal.os.Zygote$MethodAndArgsCaller.run() (Zygote.java:245)
        at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:921)
D/HwGalleryCacheManagerImpl: mIsEffect:false
D/HwRTBlurUtils: check blur style for HwPhoneWindow, themeResId : 0x7f0f01d0, context : com.example.testble4.MainActivity@5ae0945, Nhwext : 0, get Blur : disable with , null
D/HwRTBlurUtils: check blur style for HwPhoneWindow, themeResId : 0x7f0f01d0, context : com.example.testble4.MainActivity@5ae0945, Nhwext : 0, get Blur : disable with , null
D/ActivityThread: add activity client record, r= ActivityRecord{59de7a1 token=android.os.BinderProxy@1531090 {com.example.testble4/com.example.testble4.MainActivity}} token= android.os.BinderProxy@1531090
I/HwSecImmHelper: mSecurityInputMethodService is null
D/OpenGLRenderer:   HWUI Binary is  enabled
D/OpenGLRenderer: HWUI GL Pipeline
I/PressGestureDetector: onAttached begin
I/PressGestureDetector: onAttached end
I/PressGestureDetector: HiTouch restricted: AboardArea.
I/OpenGLRenderer: Initialized EGL, version 1.4
D/OpenGLRenderer: Swap behavior 2
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000
W/InputMethodManager: startInputReason = 1
W/InputMethodManager: startInputReason = 5
D/BluetoothCentralManager: popup hack completed
I/BluetoothAdapter: getBluetoothLeScanner
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
I/BluetoothCentralManager: scan started

weliem, 2021-04-26:
If you test peripheral isn't found during a scan, it must be because it is not advertising the HRS service in its advertisement packet.

You can verify this using nRF connect...

ma8ra, 2021-05-02:
Thanks for help! The problem is what you said.

closed 2021-05-02 by ma8ra