weliem/blessed-android #88: How do I get the list of peripherals?

sylque, 2021-05-19:
I want to populate a list of peripherals, so that the user can select the one he wants to connect to. 

Unfortunately, it seems the `scanForPeripheralsXXX()`+callback mechanism allows either:
- to get the first peripheral only (when calling `mCentral.stopScan()` in the callback), or 
- to get a continuous and endless loop of the same peripherals. 

What am I missing?

weliem, 2021-05-20:
ScanForPeripherals() will scan continuously and only stops when you call stopScanning()

So simply dont stop the scan...

sylque, 2021-05-20:
Thanks for your answer.
If I don't stop the scan, I get an unlimited number of duplicates:
```
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: LaserPa E7:A9:04:1C:A9:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: LaserPa E7:A9:04:1C:A9:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
I/PERIPHERAL: noname 1C:98:EC:78:A0:54
...
```
Of course I could wait a bit before stopping the scan, then remove the duplicates manually. That sounds cumbersome, though. Isn't there any other way?

sylque, 2021-05-20:
After reading a bit about it, it seems continuous scanning is the way the underlying Android BLE layer works. Sorry for the dumb question.

closed 2021-05-20 by sylque