weliem/blessed-android #89: Feature request: access to advertised data

steph643, 2021-05-24:
While debugging my BLE Android app, I felt the need to access advertised data (to check that my BLE server app was advertising correctly). Unless I'm wrong, there's not much about this in the library.

For example, `getServices()` allows to get the list of services available from a connected peripheral, but it would be nice to also have a way to get the list of advertised/exposed services from a not-connected peripheral (returned by `onDiscoveredPeripheral()`).

weliem, 2021-05-25:
The advertised data is also available and encapsulated by the ScanResult object. You get this along with the peripheral object:

`onDiscoveredPeripheral(@NotNull BluetoothPeripheral peripheral, @NotNull ScanResult scanResult)`

The ScanResult is a plain Android object, see [ScanResult](https://developer.android.com/reference/android/bluetooth/le/ScanResult)

The information about which services are advertised is in the ScanRecord:
`List<ParcelUuid> serviceUUIDs = scanResult.getScanRecord().getServiceUuids();`

steph643, 2021-05-26:
I though more about this. When I first used the library, I tried to call `getServices()` on the unconnected peripheral object. Because it returned nothing, I thought "too bad, I can't get advertised services". Maybe calling `getServices()` on an unconnected peripheral should throw a `NotConnectedPeripheral` exception instead?

weliem, 2021-05-26:
Indeed, most methods of peripheral  cannot be used unless you are connected.  But since that is rather obvious, I decided not to throw exceptions for it, although there are checks for being connected in the code.

Another reason for not using exceptions is that I think they should only be used for clear programmer errors. After being connected, a connection might drop any time. So instead of throwing exceptions for totally normal situations I decided to handle it gracefully. Otherwise your code would be full of try/catch statements....

steph643, 2021-05-26:
> But since that is rather obvious...

My point exactly: it sounds obvious at first but it's not, as `getServices()` could have returned the advertised services of a non-connected peripheral.

> After being connected, a connection might drop any time. So instead of throwing exceptions for totally normal situations I decided to handle it gracefully. Otherwise your code would be full of try/catch statements....

"Not handling" and "handling gracefully" are two different things. The user is informed of a connection drop through `onPeripheralDisconnected()`, and I think it should be his choice to manage the exceptions or not.

closed 2021-05-26 by steph643