weliem/blessed-android #91: Callbacks location

steph643, 2021-05-24:
`onConnectedPeripheral()` and `onDisconnectedPeripheral()` are in `BluetoothCentralManagerCallback`. It would sound more logical to me to put them in `BluetoothPeripheralCallback` (as `onConnected()` and `onDisconnected()`).

weliem, 2021-05-25:
Actually, when you have to deal with multiple peripherals being connected at the same time, it is much more convenient to have it all in one place. You can then deal with peripheral connections, errors and disconnects in a central place and avoid code duplications.

It is not for nothing that CoreBluetooth (by Apple) is also doing it like this...

steph643, 2021-05-26:
Code duplications? Seriously?

closed 2021-05-26 by steph643