weliem/blessed-android #95: Disable logging

MDXDave, 2021-06-21:
Can you please add the possibility to disable the logging for the library *without* disable the general Timber Logging for the whole app? Would be nice 😄

weliem, 2021-06-22:
Hi, that is not a bad idea. I will look into it.

So do I understand it correctly that you are using Timber in you application but you want to turn off the logging done by Blessed?

Or are you using a different logger?

MDXDave, 2021-06-23:
Exactly, I use Timber within my app and want to disable the logging from Blessed, without disable the logging of my app itself. 😄

weliem, 2021-06-24:
I added a function central.disableLogging()

MDXDave, 2021-06-25:
Thanks for your quick response and implementation! It works as expected. Thank you!

closed 2021-06-25 by MDXDave

agap, 2023-02-14:
Is it possible to expose the logger to the outside world? I want to be able to turn it off, but I don't use `BluetoothCentralManager` as I only need `BluetoothPeripheral`. :) So creating `BluetoothCentralManager` only to be able to turn logging off seems like an overkill 🤔 

I can add a PR ofc, but I just wanted to know your opinion on that.

And thanks for the great library - it saved me a lot of headache!