weliem/blessed-android #96: Listen the ConnectionState changes of BluetoothPeripheral via a callback

talatkuyuk, 2021-06-28:
Hi Weliem, thanks for this great package. I have started to use it.

I would like to listen connection state of BluetoothPeripheral via a callback and reflect the state in the ui. (BluetoothProfile.STATE_CONNECTED, BluetoothProfile.STATE_DISCONNECTED, BluetoothProfile.STATE_CONNECTING, BluetoothProfile.STATE_DISCONNECTING) 

BluetoothGattCallback has onConnectionStateChange method in BluetoothPeripheral class. Can you add a callback into BluetoothPeripheralCallback named onConnectionStateChanged(peripheral, connectionState) just for getting new connection state using BluetoothProfile or ConnectionState enum in the package. (of course this callback is gonna be in use if HciStatus is SUCCESS).

(The onConnectionUpdated callback uses only GattStatus not connection state, so that I am not able to use it for getting connection state of BluetoothPeripheral's gatt)

weliem, 2021-06-28:
The BluetoothManagerCallback has functions `onConnectedPeripheral` and `onDisconnectedPeripheral`. So that takes care of the CONNECTED and DISCONNECTED states. In practise that should serve your goal I think...

talatkuyuk, 2021-06-28:
Yes, both methods you mentioned serve my goal, but I would like to receive also CONNECTING status to show user the peripheral is getting  connected (via a text message or circular indicator).

Actually I can do that via state management, putting a state variable **Connecting** value, up to I receive `onConnectedPeripheral` or `onDisconnectedPeripheral`. But, think about if there is 2-3 peripherals at the same time in the screen, and each has different connection status (actually there is one peripheral in my case but can be more..). How do I maintain the connection status regarding especially CONNECTING status of peripherals? Too many boilerplate is needed.

Wouldn't be nice if we have a callback for each peripheral instance?

weliem, 2021-06-29:
I will think about this one. The Android stack never calls `onConnectionStateChanged` for **CONNECTING** or **DISCONNECTING**. I would have to make sure Blessed does that correctly instead. It is doable but kind of strange to do this while there is also `onConnectedPeripheral` and `onDisconnectedPeripheral`. Perhaps `onConnectingPeripheral` and `onDisconnectingPeripheral` would be more in line with the current callback design...

talatkuyuk, 2021-07-01:
Yes, `onConnectingPeripheral` and `onDisconnectingPeripheral` sound good.

weliem, 2021-07-03:
Callbacks have been added in release 2.0.8

talatkuyuk, 2021-07-03:
I tested the new callbacks. Both works. Thanks.

closed 2021-07-03 by talatkuyuk