weliem/blessed-android #97: Need to get device of the peripheral.

talatkuyuk, 2021-06-28:
Hi Weliem,

I need to broadcast the device of the BluetoothPeripheral instance when it is connected or disconnected. I can put the address of the peripheral into intent.putExtra, but in my case I need to put the bluetooth device into intent.putExtra. The bluetooth peripheral is not parcelable but the bluetooth device is parcaleble, so that I am able put the device into intent.putExtra. The thing that I would like to stress is that I need to get the device of the peripheral, but it is private.

`override fun onConnectedPeripheral(peripheral: BluetoothPeripheral) {
    Timber.i("connected to '%s' '%s'", peripheral.name, peripheral.address)
    broadcastUpdate(ACTION_PERIPHERAL_CONNECTED, peripheral.getDevice())
}`

`private fun broadcastUpdate(action: String, device: BluetoothDevice) {
        val intent = Intent(action)
        intent.putExtra(EXTRA_DATA, device)
        mainHandler.post { context.sendBroadcast(intent) }
 }`
 
 My proposal: Can you add a method into BluetoothPeripheral something like that:
 
 `/**
     * Get the device of the bluetooth peripheral.
     *
     * @return device of the bluetooth peripheral
     */
    @NotNull
    public BluetoothDevice getDevice() {
        return this.device; // or  return device;
    }`

weliem, 2021-06-28:
Why do you need the BluetoothDevice?

talatkuyuk, 2021-06-28:
I show some information about the peripheral (I mean about device), since BluetoothPeripheral is not parcelable I need to send BluetoothDevice via intent. I am using also Navigation package, safe arguments accept an argument if it is parcaleble or primitive data types. 

This fragment is bonded to a bound service, I can only reach to the central manager from the service after bounding get completed that is why I have to get the BluetoothDevice without waiting bounding the fragment to the service.

weliem, 2021-06-28:
There are many ways to pass information. But if you really want to do it via an Intent you could:

1. Put all the individual information elements you need in the Intent, like the mac address
2. Or pass the mac address in the Intent and then grab the BluetoothPeripheral from the BluetoothCentralManager again using `central.getPeripheral(peripheralAddress)`. This method is used in the example app that is part of this repo.

talatkuyuk, 2021-06-28:
I use jetpack navigation package and safe arguments, and I would like to pass the BluetoothDevice to another fragment via safe arguments. I can do that because the BluetoothDevice is parcelable. 

If I pass the peripheralAddress as a string, I need to use `central.getPeripheral(peripheralAddress)` as you mentioned. But the problem is that in my case, the **bluetoothHandler** and **central** **instance** live in a bound service in my application, I can only reach to the bluetoothHandler and central after the fragment is bonded to the service. I don't want to wait just to show some basic properties of the BluetoothDevice in the screen for the beginning. I want to get these information from the BluetoothDevice that coming from safe arguments.

Is there any restriction for not to getting the BluetoothDevice of the BluetoothPeripheral in the behind scene?

weliem, 2021-06-29:
Well, what you can also try is to use the `ScanResult` that you also receive in `onDiscoveredPeripheral(BluetoothPeripheral peripheral, ScanResult scanResult)`. The `ScanResult` contains the BluetoothDevice as well as more advertisement properties. And it implements Parcelable!

You can the device by doing `scanResult.getDevice()`

talatkuyuk, 2021-07-01:
Yes, I did exactly same thing while passing the argument from scanning fragment to connect fragment, using `scanResult.device`. Not only this one, but also,  I've tried to explain why I need this property.

In any case, if you provide peripheral.getDevice(), I will be satisfied. It is up to you and your decision. Thanks for the package.

weliem, 2021-07-07:
Considering that the BluetoothDevice can easily be obtained via the scanResult, I will leave it like that. 

I have looked into making BluetoothPeripheral parcelable but that is not possible since it is not a 'data object' and contains a lot of pointers to other objects.

ceduliocezar, 2021-07-16:
@talatkuyuk Not sure if this solves your problem, but you can always get access to a BluetoothDevice instance via [BluetoothAdapter.getRemoteDevice(Address)](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getRemoteDevice(java.lang.String))

```kotlin
val address = "00:11:22:33:AA:BB"

val adapter: BluetoothAdapter  = BluetoothManager.getAdapter()
val bluetoothDevice: BluetoothDevice = adapter.getRemoteDevice(address)
```

*Legal Imprint* *(Mandatory for legal reasons, sorry for that)*
*Cedulio Silva <cedulio.silva@daimler.com>, MBition GmbH, [imprint](https://github.com/Daimler/daimler-foss/blob/master/LEGAL_IMPRINT.md)*

talatkuyuk, 2021-07-23:
Thank you @ceduliocezar, I got it.

But I don't understand why the package owner @weliem does not allow `peripheral.getDevice()` and what the restriction is for not getting BluetoothDevice of the BluetoothPeripheral in behind the scene.

ceduliocezar, 2021-07-23:
Yeah, on that I also dont know.

I dont see any problem on adding a function to return the BluetoothDevice, specially due to the fact that you can do it directly from the bluetooth adapter anyway. It would be an easier way to access.

talatkuyuk, 2021-09-11:
Hi weliem, I still need to have `peripheral.getDevice()` for the reason that I have explained above.

closed 2021-12-27 by weliem