weliem/blessed-android #98: Adding multiple services

ceduliocezar, 2021-06-29:
When creating a peripheral with multiple services, is it necessary to wait for the method ``` BluetoothPeripheralManagerCallback.onServiceAdded ``` be called before adding other services?

Example

```kotlin
val peripheralManagerCallback: BluetoothPeripheralManagerCallback =
        object : BluetoothPeripheralManagerCallback() {
            override fun onServiceAdded(status: GattStatus, service: BluetoothGattService) {
 
            }
}

val  bluetoothPeripheralManager: BluetoothPeripheralManager = BluetoothPeripheralManager(
            context,
            bluetoothManager,
            peripheralManagerCallback
)

val services: List<BluetoothGattService> = listOf(bluetoothGattService1,bluetoothGattService2,bluetoothGattService3)

services.forEach {
      bluetoothPeripheralManager.add(it)
}

```

I read the documentation and it is not clear if I should wait or not. I also looked on the source code and this command(service addition) is handled by the command queue.

weliem, 2021-06-29:
You dont have to wait. They are queued up...

weliem, 2021-06-29:
I do the same as you do in the example app: https://github.com/weliem/bluetooth-server-example/blob/main/app/src/main/java/com/welie/btserver/BluetoothServer.java

ceduliocezar, 2021-07-02:
> I do the same as you do in the example app: https://github.com/weliem/bluetooth-server-example/blob/main/app/src/main/java/com/welie/btserver/BluetoothServer.java

Yes, I'm doing it the same way. I wasnt sure since it wasnt mentioned on the docs. 

Thanks.

closed 2021-07-02 by ceduliocezar