weliem/blessed-android #99: Measurement value return null in the Broadcast Receiver

FarmerBZH, 2021-06-30:
Hi,

I cloned your project and I would like to add 2 sensors.

One sensor is the WristOx2 of Nonin and the other is the O2Ring of VIATOM.

The Nonin works with your implementation PulseOximeterContinuousMeasurement but I would like to make a file on my own.

As your files, I made NoninMeasurement and O2Measurement files.

Furthermore, I completed the Bluetooth Handler and MainActivity files. 

I just made some "copy-past" and changed the UUID Value because it works in the same way.

The problem is in the onReceive function of my broadcast receiver : 

BluetoothPeripheral peripheral = getPeripheral(intent.getStringExtra(BluetoothHandler.MEASUREMENT_EXTRA_PERIPHERAL));
O2Measurement measurement = (O2Measurement) intent.getSerializableExtra(BluetoothHandler.MEASUREMENT_O2);

I always have: measurement = null

This is the same for the Nonin...

Do you have some ideas?

Lucas

FarmerBZH, 2021-06-30:
package com.welie.blessedexample;

import android.annotation.SuppressLint;

import com.welie.blessed.BluetoothBytesParser;

import java.io.Serializable;
import java.nio.ByteOrder;

import static com.welie.blessed.BluetoothBytesParser.FORMAT_SFLOAT;
import static com.welie.blessed.BluetoothBytesParser.FORMAT_UINT16;
import static com.welie.blessed.BluetoothBytesParser.FORMAT_UINT8;

public class NoninMeasurement implements Serializable {

    private final int SpO2;

    private final int pulseRate;

    public NoninMeasurement(byte[] value) {
        BluetoothBytesParser parser = new BluetoothBytesParser(value);

        //SpO2 = parser.getIntValue(FORMAT_UINT8,7, ByteOrder.LITTLE_ENDIAN);

        //pulseRate = parser.getIntValue(FORMAT_UINT16,8, ByteOrder.BIG_ENDIAN);
        SpO2 = parser.getFloatValue(FORMAT_SFLOAT,7, ByteOrder.LITTLE_ENDIAN).intValue();
        pulseRate = parser.getFloatValue(FORMAT_SFLOAT,8, ByteOrder.BIG_ENDIAN).intValue();

    }

    public int getSpO2() {
        return SpO2;
    }

    public int getPulseRate() {
        return pulseRate;
    }

    @SuppressLint("DefaultLocale")
    @Override
    public String toString() {
        if (SpO2 == 2047 || pulseRate == 2047) {
            return "invalid measurement";
        }
        return String.format("SpO2 %d%%, Pulse %d bpm", SpO2, pulseRate);
    }
}

FarmerBZH, 2021-06-30:
package com.welie.blessedexample;

import android.annotation.SuppressLint;

import com.welie.blessed.BluetoothBytesParser;

import java.io.Serializable;
import java.nio.ByteOrder;

import static com.welie.blessed.BluetoothBytesParser.FORMAT_FLOAT;
import static com.welie.blessed.BluetoothBytesParser.FORMAT_SFLOAT;
import static com.welie.blessed.BluetoothBytesParser.FORMAT_UINT8;

public class O2Measurement implements Serializable {

    private int SpO2 = 0;

    private int pulseRate = 0;

    private int battery = 0;

    public O2Measurement(byte[] value) {

        if (value.length > 2){
            SpO2 = (int) value[7] ;
            pulseRate = ((int) value[9] << 8) + (int) value[8];
            battery = (int) value[14];
        }

    }

    public int getSpO2() {
        return SpO2;
    }

    public int getPulseRate() {
        return pulseRate;
    }

    public int getBattery() {
        return battery;
    }

    @SuppressLint("DefaultLocale")
    @Override
    public String toString() {
        if (SpO2 == 2047 || pulseRate == 2047) {
            return "invalid measurement";
        }
        return String.format("SpO2 %d%%, Pulse %d bpm, Battery %d", SpO2, pulseRate, battery);
    }
}

FarmerBZH, 2021-06-30:
Extract of BluetoothHandler.java : 

In onCharacteristicUpdate():

 else if (characteristicUUID.equals(O2_MEASUREMENT_CHARACTERISTIC_UUID)) {
                O2Measurement measurement = new O2Measurement(value);
                Intent intent = new Intent(MEASUREMENT_O2);
                intent.putExtra(MEASUREMENT_O2_EXTRA, measurement);
                sendMeasurement(intent, peripheral);
                Timber.d("%s", measurement);
            }
else if (characteristicUUID.equals(NONIN_MEASUREMENT_CHARACTERISTIC_UUID)) {
                NoninMeasurement measurement = new NoninMeasurement(value);
                if (measurement.getSpO2() <= 100 && measurement.getPulseRate() <= 220) {
                    Intent intent = new Intent(MEASUREMENT_NONIN);
                    intent.putExtra(MEASUREMENT_NONIN_EXTRA, measurement);
                    sendMeasurement(intent, peripheral);
                }
                Timber.d("%s", measurement);

Extract of MainActivity.java:

private final BroadcastReceiver noninReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context context, Intent intent) {
            BluetoothPeripheral peripheral = getPeripheral(intent.getStringExtra(BluetoothHandler.MEASUREMENT_EXTRA_PERIPHERAL));
            NoninMeasurement measurement = (NoninMeasurement) intent.getSerializableExtra(BluetoothHandler.MEASUREMENT_NONIN);
            if (measurement != null) {
                measurementValue.setText(String.format(Locale.ENGLISH, "SpO2 %d%%,  Pulse %d bpm\n\nfrom %s", measurement.getSpO2(), measurement.getPulseRate(), peripheral.getName()));
            }
        }
    };

    private final BroadcastReceiver o2DataReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context context, Intent intent) {
            BluetoothPeripheral peripheral = getPeripheral(intent.getStringExtra(BluetoothHandler.MEASUREMENT_EXTRA_PERIPHERAL));
            O2Measurement measurement = (O2Measurement) intent.getSerializableExtra(BluetoothHandler.MEASUREMENT_O2);
            if (measurement != null) {
                measurementValue.setText(String.format(Locale.ENGLISH, "%d (bpm) %d (SpO2) %d (bat) from %s", measurement.getPulseRate(), measurement.getSpO2(), measurement.getBattery(), peripheral.getName()));
            }
        }
    };

weliem, 2021-06-30:
You are insert the measurement using MEASUREMENT_NONIN_EXTRA but you retrieve it using MEASUREMENT_NONIN....without the EXTRA....so that leads to null. 

Same problem for the other measurement...

FarmerBZH, 2021-07-01:
Oh shit, sorry for the stupid. question ^^
Thanks for your works it's amazing!

closed 2021-07-01 by FarmerBZH