weliem/blessed-android #100: Error CMD_STARTED for long writes

apo-d, 2021-07-02:
Hi, first of all thanks for the great library, as an iOS dev that has to fix a burning Android App, it has helped me big time.

My issue comes up mainly with Samsung devices so far, and has to do with long writes. We are using a UART-based protocol with its own delimiters. Right now, i am using "peripheral.writeCharacteristic(rxCharacteristic, encoded, WriteType.WITH_RESPONSE)", which always works perfectly for some devices. However, the Samsungs (and an HTC) seem to be overflown when a sequence of messages is sent after a bonding event (which we also need to do). In this case I am (deterministically) getting a CMD_STARTED error, which according to the HCI logs triggers an "Execute Write Request, Cancel All" (which i think does not come from your library).
I am calling the write-code above 3 times with 25, 30 and 27 bytes. It also has the same behaviour with fewer writes but longer packets and it also happens regardless of bonding in cases of even longer packets.
The MTU of the connection is limited to 23 bytes.
Something that worked in the past was the "split" of the Nordic Library (https://github.com/NordicSemiconductor/Android-BLE-Library/blob/0b956f47b299e795f24c3850646c60c71a2266ab/ble/src/main/java/no/nordicsemi/android/ble/WriteRequest.java). However there were other issues with this library. I guess this means that the Samsungs do not support long write and i would need to implement such a logic (split) for my app too? Would you think that it makes sense to add this as an option for your library? Or even better have another idea? :)
I am looking forward to your answer.

weliem, 2021-07-03:
Hi, interesting problem. Can you share a relevant piece of log so I can see exactly what you are doing? 

I don't recall having issues with Samsung phones and long writes. My main phone is a S20....but I will double-check.

The CMD_STARTED may suggest it is still doing the previous long write. I would like to see if I can reproduce your error.

apo-d, 2021-07-06:
Hi, "unfortunately" I have solved the issue and cannot send logs now, since the time schedule is kinda tight :) According to the logs, I would sometimes receive this CMD_STARTED also for the first write, but it may well have been the case that if i call 3 times writeCharacteristic almost simultaneously, something like this comes right away.

Currently I am using this solution:
- I have an outgoingBuffer (ByteArray) variable. 
- When i have a write, i call:
`val wasEmpty = outgoingBuffer.isEmpty()
                    outgoingBuffer = outgoingBuffer.plus(encoded)
                    if (wasEmpty) {
                        writeToRx(outgoingBuffer.copyOfRange(0, 20))
                    }`
- My writeToRx function is:
`private fun writeToRx(bytes: ByteArray) {
        logger.info("Writing to RX ${bytes.size} bytes")
        val peripheral = this.peripheral
        val rxCharacteristic = this.rxCharacteristic
        if (peripheral != null && rxCharacteristic != null) {
            peripheral.writeCharacteristic(rxCharacteristic, bytes, WriteType.WITHOUT_RESPONSE)
        }
    }`
- In onCharacteristicWrite i have this code:
`logger.info("onCharacteristicWrite Success with ${outgoingBuffer.size} outgoing buffer size and ${value.size} written bytes.")
            outgoingBuffer = outgoingBuffer.drop(value.size).toByteArray()
            logger.info("New outgoing buffer size is ${outgoingBuffer.size}.")
            if (outgoingBuffer.isNotEmpty()) {
                if (outgoingBuffer.size <= 20) {
                    writeToRx(outgoingBuffer)
                    logger.info("Writing ${outgoingBuffer.size} bytes to RX.")
                } else {
                    writeToRx(outgoingBuffer.copyOfRange(0,20))
                    logger.info("Writing 20 bytes to RX.")
                }
            }`

This works with all devices i have tested so far. I was wondering whether you see it fit that such a logic could be integrated via a boolean in your writeCharacteristic function, e.g. "splitWrites", where you intercept the onCharacteristicWrite and "inject" new write operations when needed. I guess this should be not only my issue, rather something that can come up in usecases where a characteristic is used as a stream.

weliem, 2021-07-07:
Ok I see you have changed your approach to avoid the issue. When I have time I will try long writes on a Samsung to see what happens.

As for you suggestion to include this in the library; since this is something that is so simple I don't think there is much value in adding it to the library. It would only work if the receiving peripheral support this behavior and this is not exact standard behavior.

closed 2021-08-19 by apo-d