weliem/blessed-android #107: onCharacteristicUpdate issue with maximum bytes

guydoron, 2021-07-21:
I have an issue with onCharacteristicUpdate where I write to BLE device with response and response is 200 bytes, 
But, I always get only 20 first bytes, I know for sure from another app that it gets 200 bytes, Maybe there is 
some specific issue/rule in blessed SDK?

weliem, 2021-07-21:
Did you use requestMtu to ask for an MTU of 203?

weliem, 2021-07-21:
...if you want to receive 200 bytes you need to ask for an MTU of 200+3 (overhead) = 203. Alternatively you can ask for an MTU of 512 and the peripheral will return the maximum value it supports. Hopefully at least 200 in your case...

ychernysheva, 2021-08-03:
@weliem But what if supported MTU (185 for iOS, for example) is bigger than average size of response (for example, 300 bytes)? In README stated that 

> BluetoothPeripheralManager class supports long reads and writes

So can I expect that peripheral, based on Blessed code, can receive this response in the form of two sequential packets?

weliem, 2021-08-03:
If you are doing a read and the response is larger than the MTU, a long read will happen but you will receive the response in 1 packet. The bluetooth stack does the long read, resulting in multiple packets, under the hood, but it also reassembles it for you so that you get the whole packet.

weliem, 2021-08-17:
Any update on this issue?

ychernysheva, 2021-08-18:
@weliem We've faced with some issues with long writes from peripheral side. Preconditions: suppose we have minimal MTU 23 and want to write message 40 bytes long. In fact our central side received long message (more than 20 bytes), but it was corrupted (i guess due to incorrect merge of parts, produced during long write process). In our specific case we need to have possibility to read and write extra long messages (even longer than 512 bytes), so we implemented our mechanism of message serialization to extend logic, used in Blessed. Luckely, now all works well for us. Example project (BleServer) was very useful to understand how to deal with the library.
Maybe you will be interested in this observation.

weliem, 2021-08-23:
Thanks for the update! Good that you got it working. Long reads/write are a bit tricky I think and not many people seem to use/support it. In general, I would avoid using them if you want to be compatible with iOS and Bluez....

closed 2021-08-23 by weliem