weliem/blessed-android #108: Samsung devices connection issues. Error Code - 133

kartic-techindustan, 2021-08-05:
We are having connection issues reported mainly on samsung devices with android version 10. 
I researched somewhat and found somework arounds like, closing Gatt and reconnecting with a delay. 
Haven't tried this. Wanted to have your input on this one first.
Thanks.

Logs: ------

```
2021-08-04 17:08:39.088 21106-21106/ D/BluetoothAdapter: STATE_ON
2021-08-04 17:08:39.095 21106-21106/ D/BluetoothGatt: connect() - device: 30:1B:97:45:DD:35, auto: false
2021-08-04 17:08:39.095 21106-21106/ D/BluetoothAdapter: isSecureModeEnabled
2021-08-04 17:08:39.097 21106-21106/ D/BluetoothGatt: registerApp() - UUID=cfafc247-dfe8-4a17-8c09-282e33660a7f
2021-08-04 17:08:39.097 21106-21106/ D/BluetoothGatt: registerApp()
2021-08-04 17:08:39.007 21106-21106/ E/action: hide.ripple
2021-08-04 17:08:39.128 21106-21182/ D/BluetoothGatt: onClientRegistered() - status=0 clientIf=14
2021-08-04 17:08:39.963 21106-21106/ E/timerCount: 2000
2021-08-04 17:08:41.034 21106-21106/ E/timerCount: 3000
2021-08-04 17:08:42.033 21106-21106/ E/timerCount: 4000
2021-08-04 17:08:43.036 21106-21106/ E/timerCount: 5000
2021-08-04 17:08:44.037 21106-21106/ E/timerCount: 6000
2021-08-04 17:08:44.160 21106-21182/ D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=14 device=30:1B:97:45:DD:35
2021-08-04 17:08:44.167 21106-21182/ D/BluetoothGatt: close()
2021-08-04 17:08:44.173 21106-21182/ D/BluetoothGatt: unregisterApp() - mClientIf=14

2021-08-04 17:08:44.193 21106-21106/ E/timerCount: 1000
2021-08-04 17:08:44.194 21106-21106/ E/action: hide.ripple
2021-08-04 17:08:44.307 21106-21106/ D/BluetoothAdapter: STATE_ON
2021-08-04 17:08:44.309 21106-21106/ D/BluetoothGatt: connect() - device: 30:1B:97:45:DD:35, auto: false
2021-08-04 17:08:44.309 21106-21106/ D/BluetoothAdapter: isSecureModeEnabled
2021-08-04 17:08:44.310 21106-21106/ D/BluetoothGatt: registerApp()
2021-08-04 17:08:44.310 21106-21106/ D/BluetoothGatt: registerApp() - UUID=d615a768-9cff-426f-88f6-aa98c53b3278
2021-08-04 17:08:44.315 21106-21182/ D/BluetoothGatt: onClientRegistered() - status=0 clientIf=14
2021-08-04 17:08:45.195 21106-21106/ E/timerCount: 2000
2021-08-04 17:08:46.196 21106-21106/ E/timerCount: 3000
2021-08-04 17:08:47.196 21106-21106/ E/timerCount: 4000
2021-08-04 17:08:48.196 21106-21106/ E/timerCount: 5000
2021-08-04 17:08:49.197 21106-21106/ E/timerCount: 6000
2021-08-04 17:08:49.338 21106-21182/ D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=14 device=30:1B:97:45:DD:35
2021-08-04 17:08:49.342 21106-21182/ D/BluetoothGatt: close()
2021-08-04 17:08:49.352 21106-21182/ D/BluetoothGatt: unregisterApp() - mClientIf=14
```

weliem, 2021-08-05:
Are you using Blessed? Doesn't look like you are....

kartic-techindustan, 2021-08-05:
i am using blessed. just removed package names and all

weliem, 2021-08-06:
On Samsung phones the timeout for a connection is 5 seconds. This gives the 133 error. Looking at your log, this is exactly what is happening. Your peripheral is either not in range or not accepting a connection

kartic-techindustan, 2021-08-06:
1. Device is in front of me. So no range issues. Also, after closing the app and retrying it again, is connects but after 2-3 tries.
2. Also Blessed uses ScanMode.LOW_LATENCY. Will ScanMode.SCAN_MODE_BALANCED help ?
3. Also, i  have gone through various forums which suggest a workaround in case of 133 error -- closing Gatt and reconnecting with a delay. Is this a good idea ??

weliem, 2021-08-06:
Well, no connection is made. Can't tell why not based on your logs. 

Blessed alreay closes gatt on failure and retries....

Can you post a longer log with Blessed logging enabled? Maybe that will show more information.

Lastly, make sure you reboot your Samsung phone to see if that solves your problems.

weliem, 2021-08-17:
Any update on this issue?

closed 2021-08-23 by kartic-techindustan