weliem/blessed-android #109: Provide handler to BluetoothPeripheral

eric-nextsense, 2021-08-06:
I noticed that when you do operations on a BluetoothPeripheral like cancelConnection(), it posts on the mainHandler (Main looper) and there is no way to change this.

Could you explain why it uses the main looper only here? Would it be possible to provide your own handler instead? If ok i do not mind doing a pull request with the change.

weliem, 2021-08-07:
On older Samsung phones it is necessary to do all GATT operations on the main thread. I guess it not necessary for modern phones but it also doesn't hurt I think. 

Is it causing you trouble?

eric-nextsense, 2021-08-09:
That is interesting to know about older Samsung phones!

I have an application where the connection should be closed correctly
before exiting, so I wrapped the callback in a Java Future. If I listen to
the future it works fine, but if I try to do a get() in the main thread to
confirm it was done then it hangs. Not a huge deal. I could just launch the
disconnect and not worry but it felt cleaner.

closed 2021-11-17 by weliem