weliem/blessed-android #110: Batch connection (autoconnect) not working after adapter is turned off/on :: issue

droideveloper1996, 2021-08-18:
Recently I have noticed that once I turn off my Bluetooth I can see callback getting fired onBluetoothAdapterStateChanged()
and I keep on waiting for the connection to happen again but seems the connection does not happen. I even tried to create a Bluetooth peripheral then creating a new batch and starting the connection even this does not work for me.
The only way to make it work is to restart my service i.e force stop my application. 

Seems like Bluetooth adapter when restarted the old object address is not refrenced.
Any Clue how to fix this.Suggestions and help would be appreciated.

  @Override
        public void onBluetoothAdapterStateChanged(int state) {
            super.onBluetoothAdapterStateChanged(state);
            if(BluetoothAdapter.STATE_ON==state){
                Log.d(TAG, "BluetoothEnabled");
                Set<String> devices = App.getRooms().keySet();
                batch.clear();
                for (String linkedMac : devices) {
                    Log.d("TAGMac", linkedMac);
                    BluetoothPeripheral peripheral = bluetoothCentralManager.getPeripheral(linkedMac);
                    batch.put(peripheral, peripheralCallback);
                }
                bluetoothCentralManager.autoConnectPeripheralsBatch(batch);
            }
            if(BluetoothAdapter.STATE_OFF==state){
                Log.d(TAG,"BluetoothDisabled");
            }
        }

weliem, 2021-08-18:
Can you add a log so I can see what is happening?

droideveloper1996, 2021-08-31:
I did somewhat this  whenever Bluetooth is turned off in onBluetoothAdapterStateChanged call back I create a fresh batch connection and this one works perfectly fine for me. If I try to use the existing BluetoothCentralManager instance it does not works but on creating new Object it works fine. I am bit skepticle how Blessed handles the previous instance.

  @Override
        public void onBluetoothAdapterStateChanged(int state) {
            super.onBluetoothAdapterStateChanged(state);

            if (BluetoothAdapter.STATE_ON == state) {
                ArrayList<BluetoothPeripheral> peripheralsList = (ArrayList<BluetoothPeripheral>) 
                bluetoothCentralManager.getConnectedPeripherals();
                for (BluetoothPeripheral pe : peripheralsList) {
                    System.out.println("This is still connected"+ pe.getAddress());
                }

                Log.d(TAG, "BluetoothEnabled");
                bluetoothCentralManager = new BluetoothCentralManager(getApplicationContext(), callback, handler);
                bluetoothCentralManager.autoConnectPeripheralsBatch(batch);

                peripheralsList = (ArrayList<BluetoothPeripheral>) bluetoothCentralManager.getConnectedPeripherals();
                for (BluetoothPeripheral pe : peripheralsList) {
                    System.out.println("This is still connected"+ pe.getAddress());
                }
//                peripheralsList.
//                Set<String> devices = App.getRooms().keySet();
//                batch.clear();
//                for (String linkedMac : devices) {
//                    Log.d("MacAddress in enabling adapter", linkedMac);
//                    BluetoothPeripheral peripheral = bluetoothCentralManager.getConnectedPeripherals(linkedMac);
//                    Log.d(TAG, "onBluetoothAdapterStateChanged: "+peripheral);;
//                    batch.put(peripheral, peripheralCallback);
//                }
//                bluetoothCentralManager.autoConnectPeripheralsBatch(batch);
            }
            if (BluetoothAdapter.STATE_OFF == state) {
                Log.d(TAG, "BluetoothDisabled");
            }
        }

weliem, 2021-08-31:
Your finding seems to suggest that something doesn't get cleaned up properly when bluetooth is turned off. I will have a look to see if I can reproduce it and fix it.

weliem, 2021-08-31:
Can you provide a log snippet for when it is not working?

weliem, 2021-09-07:
I tried reproducing the issue but I couldn't reproduce it. Please provide more information if you are still having this issue.

weliem, 2021-09-08:
closing issue due to lack of information

closed 2021-09-08 by weliem