weliem/blessed-android #111: Removing bond during ongoing connection does not always call onDisconnectedPeripheral

apo-d, 2021-08-19:
Hi, I have found this issue in a weird way, i will describe the repro steps:
- Bluetooth connection ongoing after a successful (necessary) pairing
- Delete bond from Android Settings (don't ask how this came up)
- Sometimes the onDisconnectedPeripheral will not be called, even though an automatic reconnection is triggered (automatically by your framework i think, not android)
- Sometimes the onDisconnectedPeripheral is called and actually with the code "success", which i think is also weird.

I can reproduce this easily with Samsung phones on Android 11.
My work-around so far was to do some initialisation/resetting also at the onConnectedPeripheral callback and not rely on the onDisconnected.

I would normally not even mention it, as I get that Android callbacks are sometimes "weird" and non-deterministic, but in one case where i did not get an onDisconnected, i definitely read a "close" log message and i guess that could have been used by the Framework.

In case i can help i would be happy to.

weliem, 2021-08-21:
Hi, interesting (edge) case! When a bond is 'lost', like in your case, Blessed tries to handle this. However, I never considered the scenario you describe and only thought of a bond-lost scenario while not connected. So I recognize the the steps you are describing.

When the bond is lost while not connected, the flow is:
- When the bond is lost, a disconnect is issued because there is usually no point in staying connected.
- When the disconnect completes, it retries the connection so that the bonding procedure is 're-started'. So no disconnect is reported to the app....
- If the retry also fails then a 'connectionFailed' will be reported

But in your case, it doesn't make too much sense to retry the connection and it would be better to simply report the disconnected event. That shouldn't be too hard to fix, I'll have a look in the coming days to see if I can do a fix.

weliem, 2021-08-21:
Which Samsung phone you using and which Android version? I am not able to reproduce it with my S20...

apo-d, 2021-08-21:
I am using an A40. Somehow the S20 is indeed better than A40 on the same Android.

weliem, 2021-08-21:
Ok....and what does it run, Android 11?

apo-d, 2021-08-21:
Yep

weliem, 2021-08-22:
I think I fixed in the 2.0.11 release. Can you try and see if it fixes your problem?

apo-d, 2021-08-23:
It works! Just in case something comes up regarding this, i will re-open this issue. Thanks a lot!

closed 2021-08-23 by apo-d