weliem/blessed-android #114: Long writeCharacteristic

strike76, 2021-08-31:
I'm trying to write 40kb on a characteristic.
I wrote a function to send 20 bytes at a time.
This function is part of a more complex process and often fails.
I noticed though that putting several lines of log in the onCharacteristicWrite method always works well!! 
Why does this happen? It would seem that the writing is too fast

strike76, 2021-08-31:
By setting MTU to 10 instead of 20, the operation is successful. What does it mean?

droideveloper1996, 2021-09-01:
You can send a max of 512 bytes at a time. I would suggest you negotiate MTU to max once the connection is established and then try sending data. You can define your chunk size for the long write, this function sends 300 characters at a time.
Wrap inside the async task and this would work fine.

Rather you can create a function somewhat like this. Delay(400ms) is added because of ble hardware side limitation(Processing and handling of the data may take some time) probably you are sending data too fast (which may end up with some noisy data ) and I would suggest you see hardware side logs.
I am able to send <7000 bytes of data to my device(ESP32)
  
   public void makeSpecialConfig(String config) {
        Log.d(TAG, "makeSpecialConfig Final: " + config);
        int CHUNK_SIZE = 300;
        int start = 0;
        int end = 300;
        BleConnectionManager connectionManager = BleConnectionManager.getInstance();
        while (config.length() > CHUNK_SIZE) {
            String chunks = config.substring(start, end);
            try {
                connectionManager.write(chunks);
                Thread.sleep(400);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
            config = config.substring(CHUNK_SIZE);
        }

        String finalConfig = config;
        writeHandler.postDelayed(new Runnable() {
            @Override
            public void run() {
                connectionManager.write(finalConfig);
            }
        }, 300);

    }

strike76, 2021-09-01:
@droideveloper1996 yes i use a similar functions ;) But my device accept max 20 MTU

weliem, 2021-09-02:
Can you share a log so I can see what is going on?

weliem, 2021-10-09:
Any update on this?

closed 2021-10-11 by strike76

BoonJunTan, 2022-03-11:
Hey all, can I know how to write a long characteristic using blessed library? 

I have set the MTU to 512 and tried breaking my data into 512 blocks and followed by a writeCharacteristic with chunk of data with a while loop of course. However on my receiving side it doesn't get 1 onCharacteristicWrite callback. 

Not exactly sure what I have done wrong but will really appreciate help. Thank you.

weliem, 2022-03-11:
A long write is a write where Android will break up you data in chunks of MTU-1 and then recombine it. But only up to a max of 512 because that is the max length of ancharacteristic.

So in your case, Android will transfer the 512 bytes. But it will not recombine multiple blocks of 512 bytes into a larger byte array.

BoonJunTan, 2022-03-11:
Hey @weliem. Thanks for your reply. I do understand that the max length of a characteristic is 512 given that the requestMTU is successfully set to 512.

I do also have a Windows application that is able to writeCharacteristic of more than 512 bytes (> 10k byte data) without breaking up into multiple blocks on my own as the BLE library handles that automatically and on my receiving Android application I see multiple incoming packets and the onCharacteristicWrite only get called once after all the packets arrived. 

I am just curious how I can achieve the same but instead, the writeCharacteristic is sent from an sender Android application.