weliem/blessed-android #116: Pairing dialog are duplicated at Samsung S8, Xiaomi Redme Note 10T

ermolnik, 2021-09-01:
When I try to connect some medical device: Tonometer, glucometer etc.,  the pairing dialog calls twice. And device will be successfully connected if I accept only both dialogs. I want to see only one dialog

ermolnik, 2021-09-02:
Hi, log -- https://gist.github.com/ermolnik/e772a0b794a62a8ed3e88e08eaa4b95b 
Yes, you are right, pairing dialog are called twice.
But in this application https://play.google.com/store/apps/details?id=com.ascensia.contour.us&hl=ru&gl=US all work correctly, there is only one dialog to pairing.
Do you have any idea, how can I fix it?

weliem, 2021-09-02:
Yes, just what I suspected; it comes in twice. 

I don't know how that app fixes it. I have some succesa, with other devices, to initiate the bonding yourself.

The double pairing seems to happen when it is initiated by the peripheral. But if you initiate it from the central (your app) then it may avoid the double popups.

So try calling createBond() on the peripheral instead of connectPeripheral...

weliem, 2021-09-02:
I just tried to reproduce the issue and I also get 2 pairing popups with the Contour on my Pixel 5.

I also get to 2 pairing popups when using nRF Connect. So it is not a library issue I think.

So the question remains about the original Contour app manages to avoid the issue....perhaps my suggestion fixes it. Let me know...

weliem, 2021-09-05:
I have release version 2.1.0 and that includes an update to the demo application. It issues a createBond() for the Contour and that avoids the double pairing popups.

ermolnik, 2021-09-06:
Hi, I have tried your new version, and createBond method works fine.
In this issue I told about problem with Сontour glycometer, but problems with pairing bonding dialog occur with all medical devices what i have. So, I think the problem is not in medical devices, the problem is android version and it would be more correct to check the manufacturer of the mobile device (for example, Samsung, Xiaomi, etc.) or connect to all devices with createBond()

weliem, 2021-09-07:
Indeed, I think it is an Android bug. Let me know if you convince the device manufacturers to fix the issue!

closed 2021-09-18 by weliem