weliem/blessed-android #120: onConnectionStateChange new state DISCONNECTED but BLE device is still connected

rbacevic, 2021-09-07:
Hi @weliem,

I have intermittent situation with `BluetoothGattCallback.onConnectionStateChange`. 
Status that is received is 255 (0xFF) (it is not supported in HciStatus enum, so it is converted to `UNKNOWN_STATUS_CODE`) and new state received is `DISCONNECTED`. 
Since status is not `SUCCESS` , it will call `connectionStateChangeUnsuccessful`.
Previous known state was `CONNECTED`  and services were already discovered, so next steps were to write log message: 
`BluetoothPeripheral: peripheral 'device_name' disconnected with status 'UNKNOWN_STATUS_CODE' (65535)`
and execute `completeDisconnect`.

After those steps, application thinks it is disconnected, but BLE device is still not back into advertising mode (which leads me to think that connection is still alive). Turning off then turning on Bluetooth adapter, release connection, BLE device is again in advertising mode and I was able to connect again.

So I would like to hear your thoughts about use case I described and to ask: do you have any idea what status 255 (0xFF) means ( I've been researching and found some details in Bluetooth docs about HCI statuses, but still not sure what happens).

Thanks in advance!

weliem, 2021-09-07:
Not sure what error 255 is. It could be this one:

`#define  GATT_OUT_OF_RANGE                   0xFF /* Attribute value out of range */`

The name suggests it means that the device just went out of range......does that make sense in your case?

You can see many codes [here](https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/master/stack/include/gatt_api.h) in the Android source code

rbacevic, 2021-09-07:
I was looking at that one also, and I found [this](https://stackoverflow.com/questions/63234115/write-ble-gatt-characteristic-on-android-error-status-255/63341036#63341036).

> Status 255 is GATT_OUT_OF_RANGE which means that the written value is outside of range accepted by your peripheral.

But I am pretty sure nothing is written.

Strange is that this happens few seconds after device is back in range and connected again. And the most odd thing is that callback with new state `DISCONNECTED` is executed, but BLE device still assumes it is connected. 

Thanks for your fast answer, I will continue researching about it.

weliem, 2021-09-07:
Not sure if I believe that post on SO. You get a 255 on `onConnectionStateChange` which is normally a HCI error code. In the SO post they discuss getting the error as the result of a write. So that is a GATT error. But then it is still strange because there is no official error 255, only a Android specific error code. How on earth would Android know that the value is out of range? Only the peripheral can know that....so I don't buy that story.

I still think the device is 'out of range'. Therefor you are getting a DISCONNECTED state. The other side (the BLE peripheral) might still think it is connected. Normally the connection would time out after a couple of seconds and the BLE peripheral would also be disconnected. But that all depends on the firmware code of the peripheral....

Anyway, I will add 255 as a HCI error, because apparently it is possible to get it...

weliem, 2021-09-07:
Also found this in [hci_error_code.h](https://android.googlesource.com/platform/system/bt/+/master/stack/include/hci_error_code.h)

`HCI_ERR_UNDEFINED = 0xff`

Not the most informative error. But at least it is an HCI error code from the Android source code...

closed 2021-11-23 by weliem