weliem/blessed-android #122: Bonding failed after selecting block in bluetooth pairing pop-up

kartic-techindustan, 2021-09-16:
Phone - Redmi, Model - M2101K6P
MIUI Version - MIUI GLobal 12.5.8
Android Version - 11

Redmi phone gives an option to block in pairing pop.
if user selects that, then he cannot connect to that device.
But, in os settings, he can connect and pair to that  device.

Image url -- [https://pasteboard.co/m9aTXfFmlolH.jpg](url)

Logs --

2021-09-16 17:42:33.722 6271-6271/com.bluetoothtest V/MiuiFrameworkFactory: get AllImpl object = android.common.MiuiFrameworkFactoryImpl@3fb5eab
2021-09-16 17:42:33.729 6271-6271/com.bluetoothtest W/MirrorManager: this model don't Support
2021-09-16 17:42:33.786 6271-6271/com.bluetoothtest I/Timeline: Timeline: Activity_launch_request time:160633960
2021-09-16 17:42:34.072 6271-6271/com.bluetoothtest I/BluetoothPeripheral: connect to 'EIAdv 2012000230' (30:1B:97:45:DC:F7) using transport LE
2021-09-16 17:42:34.077 6271-6271/com.bluetoothtest I/BluetoothPeripheral: peripheral '30:1B:97:45:DC:F7' is connecting
2021-09-16 17:42:34.078 6271-6271/com.bluetoothtest D/BluetoothGatt: connect() - device: 30:1B:97:45:DC:F7, auto: false
2021-09-16 17:42:34.078 6271-6271/com.bluetoothtest D/BluetoothGatt: registerApp()
2021-09-16 17:42:34.079 6271-6271/com.bluetoothtest D/BluetoothGatt: registerApp() - UUID=1b3d9ed7-6e8c-45b7-9697-4936c1b867f2
2021-09-16 17:42:34.081 6271-6358/com.bluetoothtest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2021-09-16 17:42:38.737 6271-6358/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7
2021-09-16 17:42:38.761 6271-6358/com.bluetoothtest I/BluetoothPeripheral: connected to 'EIAdv 2012000230' (NONE) in 4.7s
2021-09-16 17:42:38.765 6271-6271/com.bluetoothtest D/BluetoothPeripheral: starting bonding with 'EIAdv 2012000230' (30:1B:97:45:DC:F7)
2021-09-16 17:42:38.766 6271-6271/com.bluetoothtest D/BluetoothPeripheral: discovering services of 'EIAdv 2012000230' with delay of 0 ms
2021-09-16 17:42:38.766 6271-6271/com.bluetoothtest D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DC:F7
2021-09-16 17:42:38.782 6271-6271/com.bluetoothtest D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_CONSENT (3)
2021-09-16 17:42:38.806 6271-6271/com.bluetoothtest E/BluetoothPeripheral: bonding failed for 'EIAdv 2012000230', disconnecting device
2021-09-16 17:42:38.909 6271-6271/com.bluetoothtest I/BluetoothPeripheral: peripheral '30:1B:97:45:DC:F7' is disconnecting
2021-09-16 17:42:38.910 6271-6271/com.bluetoothtest D/BluetoothGatt: cancelOpen() - device: 30:1B:97:45:DC:F7
2021-09-16 17:42:38.915 6271-6271/com.bluetoothtest I/BluetoothPeripheral: force disconnect 'EIAdv 2012000230' (30:1B:97:45:DC:F7)
2021-09-16 17:42:38.916 6271-6358/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7
2021-09-16 17:42:38.920 6271-6358/com.bluetoothtest I/BluetoothPeripheral: disconnected 'EIAdv 2012000230' on request
2021-09-16 17:42:38.920 6271-6358/com.bluetoothtest D/BluetoothGatt: close()
2021-09-16 17:42:38.921 6271-6358/com.bluetoothtest D/BluetoothGatt: unregisterApp() - mClientIf=9
2021-09-16 17:42:38.937 6271-6271/com.bluetoothtest I/BluetoothHandler: disconnected 'EIAdv 2012000230' with status SUCCESS

weliem, 2021-09-16:
Eh, so what is the problem? If the phone blocks the pairing popup, then obviously it won't succeed....

kartic-techindustan, 2021-09-16:
Maybe user clicked that by accident, is there any way to know that device is blocked ?

weliem, 2021-09-16:
I was not aware of this functionality of the Redmi phone. Looking at the logs, it seems the pairing is immediately cancelled by the phone. So perhaps you can make the assumption that is you receive 'bonding failed' in less than 100ms after 'bonding started', it is because if the Redmi functionality?

kartic-techindustan, 2021-09-16:
I also tried a different app from playstore.
App Name - SweetBlue Toolbox
Link - https://play.google.com/store/apps/details?id=com.idevicesinc.sweetblue.toolbox&hl=en_IN&gl=US.

When i try to bond from this app, it showed that default pairing pop-up which doesn't have block feature.

weliem, 2021-09-17:
I see in your log that you are calling `central.connectPeripheral()`. If you want to bond, you need to call `central.createBond()`.

kartic-techindustan, 2021-09-20:
i was reading your medium article about Bonding - 
https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442

You have mentioned that there is no need to call createBond() and let Android handle this.

Also if i use connectPeripheral(), i receive bonding callbacks like onBondingStarted()/onBondingSucceeded()
But calling createBond(), no such callbacks are received.

Logs of createBond()--

2021-09-20 14:26:38.523 14301-14301/com.bluetoothtest V/MiuiFrameworkFactory: get AllImpl object = android.common.MiuiFrameworkFactoryImpl@8a90bee
2021-09-20 14:26:38.531 14301-14301/com.bluetoothtest W/MirrorManager: this model don't Support
2021-09-20 14:26:38.578 14301-14301/com.bluetoothtest I/Timeline: Timeline: Activity_launch_request time:278290851
2021-09-20 14:26:38.718 14301-14301/com.bluetoothtest D/BluetoothPeripheral: connecting and creating bond with 'EIAdv 2012000230'
2021-09-20 14:26:38.763 14301-14359/com.bluetoothtest I/n.bluetoothtes: ProcessProfilingInfo new_methods=1513 is saved saved_to_disk=1 resolve_classes_delay=8000
2021-09-20 14:26:38.825 14301-14301/com.bluetoothtest D/BluetoothPeripheral: starting bonding with 'EIAdv 2012000230' (30:1B:97:45:DC:F7)
2021-09-20 14:26:45.782 14301-14301/com.bluetoothtest D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_PIN (0)
2021-09-20 14:27:02.675 14301-14301/com.bluetoothtest D/BluetoothPeripheral: bonded with 'EIAdv 2012000230' (30:1B:97:45:DC:F7)
2021-09-20 14:27:02.777 14301-14301/com.bluetoothtest I/BluetoothPeripheral: connect to 'EIAdv 2012000230' (30:1B:97:45:DC:F7) using transport LE
2021-09-20 14:27:02.780 14301-14301/com.bluetoothtest I/BluetoothPeripheral: peripheral '30:1B:97:45:DC:F7' is connecting
2021-09-20 14:27:02.781 14301-14301/com.bluetoothtest D/BluetoothGatt: connect() - device: 30:1B:97:45:DC:F7, auto: false
2021-09-20 14:27:02.781 14301-14301/com.bluetoothtest D/BluetoothGatt: registerApp()
2021-09-20 14:27:02.782 14301-14301/com.bluetoothtest D/BluetoothGatt: registerApp() - UUID=bec8cf8e-0075-482c-8415-bb4313a11032
2021-09-20 14:27:02.784 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2021-09-20 14:27:02.790 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7
2021-09-20 14:27:02.794 14301-14623/com.bluetoothtest I/BluetoothPeripheral: connected to 'EIAdv 2012000230' (BONDED) in 0.0s
2021-09-20 14:27:02.795 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7
2021-09-20 14:27:02.796 14301-14301/com.bluetoothtest D/BluetoothPeripheral: discovering services of 'EIAdv 2012000230' with delay of 0 ms
2021-09-20 14:27:02.796 14301-14301/com.bluetoothtest D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DC:F7
2021-09-20 14:27:02.796 14301-14623/com.bluetoothtest I/BluetoothPeripheral: connected to 'EIAdv 2012000230' (BONDED) in 0.0s
2021-09-20 14:27:02.799 14301-14301/com.bluetoothtest D/BluetoothPeripheral: discovering services of 'EIAdv 2012000230' with delay of 0 ms
2021-09-20 14:27:02.800 14301-14301/com.bluetoothtest D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DC:F7
2021-09-20 14:27:14.042 14301-14623/com.bluetoothtest D/BluetoothGatt: onConnectionUpdated() - Device=30:1B:97:45:DC:F7 interval=6 latency=0 timeout=500 status=0
2021-09-20 14:27:14.054 14301-14623/com.bluetoothtest D/BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2021-09-20 14:27:14.357 14301-14623/com.bluetoothtest D/BluetoothGatt: onSearchComplete() = Device=30:1B:97:45:DC:F7 Status=0
2021-09-20 14:27:14.362 14301-14623/com.bluetoothtest I/BluetoothPeripheral: discovered 6 services for 'EIAdv 2012000230'
2021-09-20 14:27:14.377 14301-14301/com.bluetoothtest I/BluetoothHandler: connected to 'EIAdv 2012000230'
2021-09-20 14:27:14.801 14301-14623/com.bluetoothtest D/BluetoothGatt: onConnectionUpdated() - Device=30:1B:97:45:DC:F7 interval=306 latency=4 timeout=600 status=0
2021-09-20 14:27:14.803 14301-14623/com.bluetoothtest D/BluetoothPeripheral: connection parameters: interval=382.5ms latency=4 timeout=6s

kartic-techindustan, 2021-09-29:
> i was reading your medium article about Bonding - https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442
> 
> You have mentioned that there is no need to call createBond() and let Android handle this.
> 
> Also if i use connectPeripheral(), i receive bonding callbacks like onBondingStarted()/onBondingSucceeded() But calling createBond(), no such callbacks are received.
> 
> Logs of createBond()--
> 
> 2021-09-20 14:26:38.718 14301-14301/com.bluetoothtest D/BluetoothPeripheral: connecting and creating bond with 'EIAdv 2012000230' 2021-09-20 14:26:38.763 14301-14359/com.bluetoothtest I/n.bluetoothtes: ProcessProfilingInfo new_methods=1513 is saved saved_to_disk=1 resolve_classes_delay=8000 2021-09-20 14:26:38.825 14301-14301/com.bluetoothtest D/BluetoothPeripheral: starting bonding with 'EIAdv 2012000230' (30:1B:97:45:DC:F7) 2021-09-20 14:26:45.782 14301-14301/com.bluetoothtest D/BluetoothPeripheral: pairing request received: PAIRING_VARIANT_PIN (0) 2021-09-20 14:27:02.675 14301-14301/com.bluetoothtest D/BluetoothPeripheral: bonded with 'EIAdv 2012000230' (30:1B:97:45:DC:F7) 2021-09-20 14:27:02.777 14301-14301/com.bluetoothtest I/BluetoothPeripheral: connect to 'EIAdv 2012000230' (30:1B:97:45:DC:F7) using transport LE 2021-09-20 14:27:02.780 14301-14301/com.bluetoothtest I/BluetoothPeripheral: peripheral '30:1B:97:45:DC:F7' is connecting 2021-09-20 14:27:02.781 14301-14301/com.bluetoothtest D/BluetoothGatt: connect() - device: 30:1B:97:45:DC:F7, auto: false 2021-09-20 14:27:02.781 14301-14301/com.bluetoothtest D/BluetoothGatt: registerApp() 2021-09-20 14:27:02.782 14301-14301/com.bluetoothtest D/BluetoothGatt: registerApp() - UUID=bec8cf8e-0075-482c-8415-bb4313a11032 2021-09-20 14:27:02.784 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9 2021-09-20 14:27:02.790 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7 2021-09-20 14:27:02.794 14301-14623/com.bluetoothtest I/BluetoothPeripheral: connected to 'EIAdv 2012000230' (BONDED) in 0.0s 2021-09-20 14:27:02.795 14301-14623/com.bluetoothtest D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DC:F7 2021-09-20 14:27:02.796 14301-14301/com.bluetoothtest D/BluetoothPeripheral: discovering services of 'EIAdv 2012000230' with delay of 0 ms 2021-09-20 14:27:02.796 14301-14301/com.bluetoothtest D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DC:F7 2021-09-20 14:27:02.796 14301-14623/com.bluetoothtest I/BluetoothPeripheral: connected to 'EIAdv 2012000230' (BONDED) in 0.0s 2021-09-20 14:27:02.799 14301-14301/com.bluetoothtest D/BluetoothPeripheral: discovering services of 'EIAdv 2012000230' with delay of 0 ms 2021-09-20 14:27:02.800 14301-14301/com.bluetoothtest D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DC:F7 2021-09-20 14:27:14.042 14301-14623/com.bluetoothtest D/BluetoothGatt: onConnectionUpdated() - Device=30:1B:97:45:DC:F7 interval=6 latency=0 timeout=500 status=0 2021-09-20 14:27:14.054 14301-14623/com.bluetoothtest D/BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s 2021-09-20 14:27:14.357 14301-14623/com.bluetoothtest D/BluetoothGatt: onSearchComplete() = Device=30:1B:97:45:DC:F7 Status=0 2021-09-20 14:27:14.362 14301-14623/com.bluetoothtest I/BluetoothPeripheral: discovered 6 services for 'EIAdv 2012000230' 2021-09-20 14:27:14.377 14301-14301/com.bluetoothtest I/BluetoothHandler: connected to 'EIAdv 2012000230' 2021-09-20 14:27:14.801 14301-14623/com.bluetoothtest D/BluetoothGatt: onConnectionUpdated() - Device=30:1B:97:45:DC:F7 interval=306 latency=4 timeout=600 status=0 2021-09-20 14:27:14.803 14301-14623/com.bluetoothtest D/BluetoothPeripheral: connection parameters: interval=382.5ms latency=4 timeout=6s

@weliem

weliem, 2021-09-29:
Indeed, normally speaking there is no need to call createBond() yourself. But since your phone is blocking the pairing popup I was suggesting it as a 'workaround'. So it is perfectly allowed in the BLE standard to initiate bonding yourself and I was hoping that in that case the popup wouldn't be blocked. Because you say it is not blocked when using Sweetblue...

So are you getting the pairing popup when using createBond()?

As for the callbacks, you should receive them if you passed a valid callback object.

kartic-techindustan, 2021-09-29:
Have a write a workaround for redmi devices.

Thanks!!

closed 2021-09-29 by kartic-techindustan