weliem/blessed-android #124: SCAN_FAILED_ALREADY_STARTED Getting error on Scan devices

shruti-techindustan, 2021-10-14:
Initially scanning is working. After that I disable bluetooth and enable bluetooth again then got error message SCANNING ALREADY STARTED. It only works until I kill the app and open the app.  How to fix this problem when app is already opened.

weliem, 2021-10-20:
What phone are you using? Perhaps it is a phone specific issue....

kartic-techindustan, 2021-10-21:
This also occurs in example provided by you.

**Tested on Phones** -- 
Pixel - 4A - Android12
Redmi - Android10

**Logs** -- 

2021-10-21 16:57:32.622 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning started
2021-10-21 16:57:32.641 11294-11294/com.welie.blessedexample I/BluetoothCentralManager: scan started
2021-10-21 16:57:39.169 11294-11294/com.welie.blessedexample I/BluetoothHandler: bluetooth adapter changed state to 13
2021-10-21 16:57:39.170 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning bluetooth changed
2021-10-21 16:57:39.315 11294-11294/com.welie.blessedexample I/BluetoothHandler: bluetooth adapter changed state to 10
2021-10-21 16:57:39.315 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning bluetooth changed
2021-10-21 16:57:43.227 11294-11294/com.welie.blessedexample I/BluetoothHandler: bluetooth adapter changed state to 11
2021-10-21 16:57:43.227 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning bluetooth changed
2021-10-21 16:57:43.519 11294-11294/com.welie.blessedexample I/BluetoothHandler: bluetooth adapter changed state to 12
2021-10-21 16:57:43.519 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning bluetooth changed
2021-10-21 16:57:44.529 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning started
2021-10-21 16:57:44.530 11294-11294/com.welie.blessedexample I/BluetoothCentralManager: scan started
2021-10-21 16:57:44.531 11294-11294/com.welie.blessedexample E/BluetoothCentralManager: scan failed with error code 1 (ALREADY_STARTED)
2021-10-21 16:57:44.532 11294-11294/com.welie.blessedexample I/BluetoothHandler: scanning failed with error ALREADY_STARTED