weliem/blessed-android #125: 2 pairing dialogs on Nonin 3230 first connection

boyks, 2021-11-12:
Device: Pixel 4a
OS: Android 12

When connecting to the Nonin 3230 device for the first time, I need to confirm pairing 2 times on the system pop-up dialog.
As far as I understand, the Nonin device requires bonding for a secure connection, thus requesting pairing with the device. But why 2 same dialogs?

weliem, 2021-11-12:
It's an Android bug. It happens on some phone but not all. 

The workaround is to call createBond() instead of connectGatt() if the device is not bonded yet. 

In the example app of this library, this is done for the Contour glucose meter.

weliem, 2021-11-23:
Any progress on this?

boyks, 2021-11-23:
Yeah.
Tried implementing my own stack based on your example, and indeed calling createBond() fixes the problem, but not calling it _instead_ connectGatt(), but just _before_ it, then listening to bond changes.

Also have limited this workaround to Google (Pixel) and Samsung devices, not sure though if it's enough - will wait for responses from users.

closed 2021-11-23 by boyks