weliem/blessed-android #126: setCharacteristicNotification  not working after close connection

shruti-techindustan, 2021-11-23:
Not able to enable Notification for Characteristic when making connection after cancel connection 

2021-11-23 17:37:54.700 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp()
2021-11-23 17:37:54.700 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp() - UUID=c07491d0-ce20-4c1c-b553-2f8c6b969b70
2021-11-23 17:37:54.704 29960-29988/com.emerald.ems.customer D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2021-11-23 17:37:55.259 29960-29988/com.emerald.ems.customer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=48:00:00:01:38:1A
2021-11-23 17:37:55.305 29960-29960/com.emerald.ems.customer D/BluetoothGatt: discoverServices() - device: 48:00:00:01:38:1A
2021-11-23 17:37:55.313 29960-29988/com.emerald.ems.customer D/BluetoothGatt: onSearchComplete() = Device=48:00:00:01:38:1A Status=0
2021-11-23 17:37:55.401 29960-29960/com.emerald.ems.customer D/BluetoothGatt: configureMTU() - device: 48:00:00:01:38:1A mtu: 517
2021-11-23 17:37:55.598 29960-29988/com.emerald.ems.customer D/BluetoothGatt: onConfigureMTU() - Device=48:00:00:01:38:1A mtu=23 status=0
2021-11-23 17:37:55.605 29960-29960/com.emerald.ems.customer D/BluetoothGatt: requestConnectionPriority() - params: 1
2021-11-23 17:37:55.912 29960-29988/com.emerald.ems.customer D/BluetoothGatt: onConnectionUpdated() - Device=48:00:00:01:38:1A interval=12 latency=0 timeout=500 status=0
2021-11-23 17:37:56.205 29960-29960/com.emerald.ems.customer D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002b10-0000-1000-8000-00805f9b34fb enable: true
2021-11-23 17:37:56.406 29960-29993/com.emerald.ems.customer D/BluetoothGatt: onConnectionUpdated() - Device=48:00:00:01:38:1A interval=312 latency=4 timeout=600 status=0
2021-11-23 17:43:52.235 29960-29960/com.emerald.ems.customer D/BluetoothGatt: cancelOpen() - device: 48:00:00:01:38:1A
2021-11-23 17:43:52.246 29960-29993/com.emerald.ems.customer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=48:00:00:01:38:1A
2021-11-23 17:43:52.254 29960-29993/com.emerald.ems.customer D/BluetoothGatt: close()
2021-11-23 17:43:52.264 29960-29993/com.emerald.ems.customer D/BluetoothGatt: unregisterApp() - mClientIf=9
2021-11-23 17:43:53.715 29960-29960/com.emerald.ems.customer D/BluetoothGatt: connect() - device: 30:1B:97:45:DD:98, auto: false
2021-11-23 17:43:53.716 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp()
2021-11-23 17:43:53.717 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp() - UUID=3478dc30-1132-45a0-b567-5122f87b5a72
2021-11-23 17:43:53.729 29960-31181/com.emerald.ems.customer D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2021-11-23 17:43:58.759 29960-30478/com.emerald.ems.customer D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=30:1B:97:45:DD:98
2021-11-23 17:43:58.766 29960-30478/com.emerald.ems.customer D/BluetoothGatt: close()
2021-11-23 17:43:58.772 29960-30478/com.emerald.ems.customer D/BluetoothGatt: unregisterApp() - mClientIf=9
2021-11-23 17:43:58.923 29960-29960/com.emerald.ems.customer D/BluetoothGatt: connect() - device: 30:1B:97:45:DD:98, auto: false
2021-11-23 17:43:58.925 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp()
2021-11-23 17:43:58.926 29960-29960/com.emerald.ems.customer D/BluetoothGatt: registerApp() - UUID=1c80e579-2777-4c94-9019-1731d9a184aa
2021-11-23 17:43:58.937 29960-30478/com.emerald.ems.customer D/BluetoothGatt: onClientRegistered() - status=0 clientIf=9
2021-11-23 17:44:00.489 29960-30478/com.emerald.ems.customer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=30:1B:97:45:DD:98
2021-11-23 17:44:00.530 29960-29960/com.emerald.ems.customer D/BluetoothGatt: discoverServices() - device: 30:1B:97:45:DD:98
2021-11-23 17:44:00.544 29960-30478/com.emerald.ems.customer D/BluetoothGatt: onSearchComplete() = Device=30:1B:97:45:DD:98 Status=0
2021-11-23 17:44:01.806 29960-30478/com.emerald.ems.customer D/BluetoothGatt: onConnectionUpdated() - Device=30:1B:97:45:DD:98 interval=312 latency=4 timeout=600 status=0

weliem, 2021-11-23:
In the log, it doesn't show that you are doing a setNotify the second time. Not even the other operations like requesting an MTU is showing.

Is the log incomplete or did it not really happen?

weliem, 2021-12-28:
Any update on this issue?

weliem, 2022-01-21:
Closing due to lack of information

closed 2022-01-21 by weliem