weliem/blessed-android #127: Sample app

decharre, 2021-11-30:
Hi, I am pretty new to Android Studio.
I am trying to get confident with this library, however, I only get a "waiting for measurement" homepage, where I was hoping to get a list of BLE device which I could select to read out their services and characteristics.

Can someone give me a tip on how to get started with an easy example ? 

I am trying to read out blood pressure measurements by bluetooth:

bluetooth device : Medisana BU 570 connect

Bluetooth device : FC:45:C3:A1:92:1B
Service 00001810-0000-1000-8000-00805f9b34fb
characteristic : 00002a35-0000-1000-8000-00805f9b34fb: INDICATABLE

(With an nrf scanner app, I can see the device, connect to it, and I can see all different blood pressure measurements, so I know it must be possible.)

Thanks in advance

weliem, 2021-11-30:
Just run the sample app and take a measurement. After the measurement is taken your device will start advertising and the app should automatically connect to it.

decharre, 2021-11-30:
Hi Weliem,  we are trying, but it looks like the app is not connecting to our BP. When we try with the nrf sniffer, we get the value, so it looks like this must be possible. Any further tips you can give ?
Thanks

nRF Connect, 2021-11-21
LS BPM (FC:45:C3:A1:92:1B)
I	16:54:03.336	[Server] Server started
V	16:54:03.344	Heart Rate (0x180D)
- Heart Rate Measurement [N] (0x2A37)
   Client Characteristic Configuration (0x2902)
- Body Sensor Location [R] (0x2A38)
- Heart Rate Control Point [W] (0x2A39)
Unknown Service (0000aaa0-0000-1000-8000-aabbccddeeff)
- Unknown Characteristic [N R] (0000aaa1-0000-1000-8000-aabbccddeeff)
   Client Characteristic Configuration (0x2902)
   Unknown Descriptor (0000aab0-0000-1000-8000-aabbccddeeff)
   Characteristic User Description (0x2901)
   Characteristic Presentation Format (0x2904)
- Unknown Characteristic [I W WNR] (0000aaa2-0000-1000-8000-aabbccddeeff)
   Client Characteristic Configuration (0x2902)
User Data (0x181C)
- First Name [R W] (0x2A8A)
- Last Name [R W] (0x2A90)
- Gender [R W] (0x2A8C)
Unknown Service (00002a35-0000-1000-8000-00805f9b34fb)
V	16:54:03.619	Connecting to FC:45:C3:A1:92:1B...
D	16:54:03.619	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	16:54:33.638	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	16:54:33.638	[Server] Device disconnected
D	16:54:33.651	[Callback] Connection state changed with status: 133 and new state: DISCONNECTED (0)
E	16:54:33.652	Error 133 (0x85): GATT ERROR
I	16:54:33.652	Disconnected
D	16:54:42.274	gatt.close()
D	16:54:42.275	wait(200)
V	16:54:42.477	Connecting to FC:45:C3:A1:92:1B...
D	16:54:42.478	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	16:54:56.151	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:54:56.152	[Server] Device with address FC:45:C3:A1:92:1B connected
D	16:54:56.158	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:54:56.158	Connected to FC:45:C3:A1:92:1B
D	16:54:56.177	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	16:54:56.181	Discovering services...
D	16:54:56.181	gatt.discoverServices()
I	16:54:56.448	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	16:54:56.916	[Callback] Services discovered with status: 0
I	16:54:56.916	Services discovered
V	16:54:56.938	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Privacy Flag [R W] (0x2A02)
- Reconnection Address [W] (0x2A03)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Device Information (0x180A)
- System ID [R] (0x2A23)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
- Software Revision String [R] (0x2A28)
- Manufacturer Name String [R] (0x2A29)
- IEEE 11073-20601 Regulatory Certification Data List [R] (0x2A2A)
- PnP ID [R] (0x2A50)
Unknown Service (0000fff0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W] (0000fff1-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
- Unknown Characteristic [R] (0000fff2-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
- Unknown Characteristic [R W] (0000fff3-0000-1000-8000-00805f9b34fb)
   Characteristic User Description (0x2901)
- Unknown Characteristic [N] (0000fff4-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
[... 21 lines cut ...]
D	16:54:56.940	gatt.setCharacteristicNotification(00002a36-0000-1000-8000-00805f9b34fb, true)
D	16:54:56.941	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
I	16:54:57.017	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 5000ms)
I	16:54:57.342	Connection parameters updated (interval: 90.0ms, latency: 0, timeout: 6000ms)
V	16:55:02.584	Enabling indications for 00002a35-0000-1000-8000-00805f9b34fb
D	16:55:02.585	gatt.setCharacteristicNotification(00002a35-0000-1000-8000-00805f9b34fb, true)
D	16:55:02.586	gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0200)
I	16:55:02.745	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 02-00
A	16:55:02.745	"Indications enabled" sent
V	16:55:02.752	Indications enabled for 00002a35-0000-1000-8000-00805f9b34fb
I	16:55:03.660	Indication received from 00002a35-0000-1000-8000-00805f9b34fb, value: (0x) 1E-84-00-4F-00-60-00-E0-07-01-0F-13-0F-00-45-00-00-B8-00
A	16:55:03.660	"Systolic: 132.0 mmHg
Diastolic: 79.0 mmHg
Mean AP: 96.0 mmHg
Timestamp: 19:15:00 14.02.2016
Pulse: 69.0 bpm
User ID: 0
Pulse rate range: Reserved for future use 
Improper measurement position" received
I	16:55:04.846	Indication received from 00002a35-0000-1000-8000-00805f9b34fb, value: (0x) 1E-80-00-44-00-58-00-E0-07-01-0F-13-0E-00-3D-00-00-98-00
A	16:55:04.846	"Systolic: 128.0 mmHg
Diastolic: 68.0 mmHg
Mean AP: 88.0 mmHg
Timestamp: 19:14:00 14.02.2016
Pulse: 61.0 bpm

weliem, 2021-11-30:
It depends on what it advertises...can you check in nrfConnect?

Alternatively, you can modify the code to scan for peripherals with the name "LS BPM"

decharre, 2021-12-01:
so, we would like to use the library to 

1) connect to LS BPM (FC:45:C3:A1:92:1B)

2) turn on CharacteristicNotification for 00002a35-0000-1000-8000-00805f9b34fb

16:55:02.584 Enabling indications for 00002a35-0000-1000-8000-00805f9b34fb
D 16:55:02.585 gatt.setCharacteristicNotification(00002a35-0000-1000-8000-00805f9b34fb, true)
D 16:55:02.586 gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0200)
I 16:55:02.745 Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 02-00
A 16:55:02.745 "Indications enabled" sent

3)  receive values from 00002a35-0000-1000-8000-00805f9b34fb

4) interpret the value: coming from this characteristic : example : (0x) 1E-84-00-4F-00-60-00-E0-07-01-0F-13-0F-00-45-00-00-B8-00

 "Systolic: 132.0 mmHg
Diastolic: 79.0 mmHg
Mean AP: 96.0 mmHg
Timestamp: 19:15:00 14.02.2016
Pulse: 69.0 bpm
User ID: 0
Pulse rate range: Reserved for future use
Improper measurement position" received

could you let me know or provide a sample code to connect to a BLE device, set Notifications on for a certain characteristic ?

thank you

weliem, 2021-12-01:
The sample app provides all the basics you need. I am sure you can figure out to make the necessary changes...

decharre, 2021-12-07:
Hi Weliem,
I am afraid we cannot get it to work. Couldn't you provide me a more advanced sample app to read out the characteristics of a BLE peripheral ? Much appreciated, Dirk

closed 2021-12-27 by weliem

r-arteaga, 2023-03-24:
Same here, doesn't detect my Polar H10. Only the "waiting for measurement..." text