weliem/blessed-android #129: BluetoothPeripheralManager is incorrectly chopping long value

ceduliocezar, 2022-01-03:
TLDR; (BluetoothPeripheralManager)[https://github.com/weliem/blessed-android/blob/69b44ff18c6899050c204a820959d3a1b4edd93b/blessed/src/main/java/com/welie/blessed/BluetoothPeripheralManager.java#L178] seems to be incorrectly chopping long values when BluetoothDevice.TRANSPORT_BREDR is used as transport. If BluetoothDevice.TRANSPORT_LE is used the long read works fine .

I ran several tests locally with two android phones and an iphone(central role). My observation is that when BluetoothDevice.TRANSPORT_BREDR is used the OS stack does not issue subsequent calls incrementing the offset. Both Android OS and iOS have this behavior.

Scenario
- Android PhoneA runs peripheral server with blessed-android.
- Android PhoneB runs central with blessed-android.

Steps to reproduce the problem:
- Android PhoneA runs Peripheral role with 1 service and 1 characteristic with long value ( it has to be longer that default mtu size)

``` 
val characteristic = BluetoothGattCharacteristic(CHARACTERISTIC_ID, BluetoothGattCharacteristic.PROPERTY_READ, BluetoothGattCharacteristic.PERMISSION_READ_ENCRYPTED)

characteristic.value = "Lorem ipsum dolor sit amet, consectetur adipiscing elit" 
```
- Android PhoneB connects to PhoneA peripheral using BluetoothDevice.TRANSPORT_BREDR.
- Service discovery happens.
- PhoneB issues a read request to characteristic.
- PhoneA responds to the request chopping the end of the byte array.
- PhoneB receives part of the byte.

<sub>Legal Imprint (Mandatory for legal reasons, sorry for that)</sub>
<sub>Cedulio Silva <cedulio.silva@daimler.com>, MBition GmbH, [imprint](https://github.com/Daimler/daimler-foss/blob/master/LEGAL_IMPRINT.md)</sub>

ceduliocezar, 2022-01-03:
I'm working on the fix and I will create a PR as soon I write the tests.

ceduliocezar, 2022-01-04:
PR created https://github.com/weliem/blessed-android/pull/130

weliem, 2022-01-13:
Merged PR and released version 2.1.4

I wonder if the same issue also occurs when doing long reads on descriptors.....because that uses the same implementation.

ceduliocezar, 2022-01-14:
@weliem I did a testing reading long descriptor values and indeed it has the same problem

weliem, 2022-01-21:
Ok, applied the same fix to descriptors in release 2.1.5.

Can you test and confirm it now also works for descriptors?