weliem/blessed-android #139: Looper.getMainLooper()

WanManito, 2022-04-02:
The callback of the whole API is executed in the main thread. In some scenarios, when used in combination with wait (2000), the main thread will be blocked for two seconds when waiting for the message to return. The effect I want is that I send data in the main thread and receive data in the sub thread. I don't know whether the existing can do it. Thank you. When common TCP communication uses data flow to obtain data, it is obtained in the sub thread in a circular way

weliem, 2022-04-14:
Hi,

You can control on which thread you get your callbacks. You pass a Handler object when the library is initialised:

`central = new BluetoothCentralManager(context, bluetoothCentralManagerCallback, new Handler());`

So you do this call on the main thread, the Handler will run on the main thread.
If you want to use a separate thread you can do something like this:

```
syncToServerHandlerThread = new HandlerThread("SyncToServerHandlerThread");
syncToServerHandlerThread.start();
syncToServerHandler = new Handler(syncToServerHandlerThread.getLooper());
central = new BluetoothCentralManager(context, bluetoothCentralManagerCallback, syncToServerHandler);
```

WanManito, 2022-04-15:
Hello, I'm glad to receive your reply. According to the "handlerthread" example you gave me, I sent an instruction in the main thread, followed by a line of wait (3000), which still blocked the handlerthread thread. I don't know where there is a problem
[image]

What I expect is that before the end of these three seconds, the callback of data reception will reply. In fact, the effect is to wait for the end of three seconds before the callback receives a reply.

weliem, 2022-04-16:
When you use 'wait', you are blocking the thread you are on at that moment. Probably you are on the same thread as the thread you passed to Blessed. So it is logical that it doesn't work.

By the way, I don't understand what you are trying to do. Why do a 'wait'?

WanManito, 2022-04-18:
Hello, wait is because the received data is asynchronous. I want to assemble a complete protocol data during this wait and return it to the main thread. Then wait also has the function of timeout. Now, I don't know why handlerthread still blocks the main thread. I've tried for a long time

weliem, 2022-04-27:
You are not doing it in the right way. Don't use wait in your code. If you prefer a synchronous style of programming, then take a look at [the coroutines version of this library](https://github.com/weliem/blessed-android-coroutines)

That will allow you to do things like:

```
peripheral.getCharacteristic(DIS_SERVICE_UUID, MANUFACTURER_NAME_CHARACTERISTIC_UUID)?.let {
    val manufacturerName = peripheral.readCharacteristic(it).asString()
    Timber.i("Received: $manufacturerName")
}

val model = peripheral.readCharacteristic(DIS_SERVICE_UUID, MODEL_NUMBER_CHARACTERISTIC_UUID).asString()
Timber.i("Received: $model")
```

closed 2022-05-03 by weliem