weliem/blessed-android #140: peripheralCallback issue in BluetoothCentralManageCallback

EddieXu123, 2022-04-13:
Hi there! I am really new to Android Development. I'm trying to use Blessed for my project to connect to a bluetooth device (my android app should be able to discover the bluetooth low energy device, and the bluetooth LE device is already discoverable [tested with LightBlue]). 

I used the first bit of code in the Readme:
`private final BluetoothCentralManagerCallback bluetoothCentralManagerCallback = new BluetoothCentralManagerCallback() {
        @Override
        public void onDiscoveredPeripheral(BluetoothPeripheral peripheral, ScanResult scanResult) {
            central.stopScan();
            central.connectPeripheral(peripheral, peripheralCallback);
        }
};`

but am getting an error with the peripheralCallback parameter in central.connectPeripheral (cannot resolve symbol). I was just wondering how I should approach this? Thank you!

EddieXu123, 2022-04-13:
Is this supposed to be another function after the device is discovered? I used central.scanForPeripheralsWithNames with the UUID of the device, I was hoping to somehow just log the device name after my app connects to it (for now). Thanks!

weliem, 2022-04-14:
Hi,

You can control on which thread you get your callbacks. You pass a Handler object when the library is initialised:

`central = new BluetoothCentralManager(context, bluetoothCentralManagerCallback, new Handler());`

So you do this call on the main thread, the Handler will run on the main thread.
If you want to use a separate thread you can do something like this:

```
syncToServerHandlerThread = new HandlerThread("SyncToServerHandlerThread");
syncToServerHandlerThread.start();
syncToServerHandler = new Handler(syncToServerHandlerThread.getLooper());
central = new BluetoothCentralManager(context, bluetoothCentralManagerCallback, syncToServerHandler);
```

closed 2022-04-19 by weliem