weliem/blessed-android #141: How do you call onCharacteristicUpdate?

EddieXu123, 2022-04-26:
Hi there! Thanks for the awesome library and for your Medium tutorials! I've been able to connect to my peripheral using the mac address and am hoping to write/read to my bluetooth device. I am following the tutorial and am able to access my peripheral. However, when I do `peripheral.readCharacteristic(onOffServ, onOffChar);` with my serviceUUID and characteristicUUID, I see this returns a boolean. I am hoping to read the byte[] information of the readCharacteristic function and see that you say it is in onCharacteristicUpdate, but I was wondering how I can call that function to return maybe a byte[]? Thank you! Here is my code if that helps! 

```
        String lampiMacAddress = "..."; //MacAddress (I'm able to get something in line 4)
        UUID onOffServ = UUID.fromString("0001A7D3-D8A4-4FEA-8174-1736E808C066");
        UUID onOffChar = UUID.fromString("0004A7D3-D8A4-4FEA-8174-1736E808C066");
        BluetoothPeripheral peripheral = central.getPeripheral(lampiMacAddress); 

        boolean out = peripheral.readCharacteristic(onOffServ, onOffChar);
        Log.d("Result", out + "");

        Log.d("Result", peripheral.getName()); // Gets correct device 
```

So I am calling the readCharacteristic but was hoping to be able to get what the characteristic info is. Thank you for your help and awesome library!

EddieXu123, 2022-04-26:
Also, I'm wondering what an example of writing would look like? This is my bluetooth device (I believe the ServiceUUID and CharacteristicUUID are correct in my above code)? 
[image]

I'm trying to do something like this to write: 
```
byte[] toWrite = new byte[]{0}; // turns device off
BluetoothGattCharacteristic onOffCharacteristic = peripheral.getCharacteristic(onOffServ, onOffChar);
peripheral.writeCharacteristic(currentTimeCharacteristic, toWrite, WriteType.WITH_RESPONSE);
```

This is all in my onCreate (Should I put it elsewhere?). I'm hoping to simply turn my device off when I run this code. However, peripheral.writeCharacteristic seems to be pausing to run of the rest of the code (and the device isn't turning off) Thanks!

weliem, 2022-04-27:
This library using callbacks so you need to set up a callback object that implements onCharacteristicUpdate. That will be called when new data comes in.

But it seems you are preferring a more synchronous styles of programming. Have a look at the : [coroutines version of this library](https://github.com/weliem/blessed-android-coroutines). That will allow you do a read of a characteristic and get the bytes immediately.

closed 2022-07-05 by weliem