weliem/blessed-android #144: Support for Android 13

MDXDave, 2022-06-15:
When I try to use my app on Android 13 Beta 3 (Google Pixel 6), it won't work, because the `onConnectionUpdated` callback is never called and thus my app will not recognize that the bluetooth device is connected with the smartphone/app.

Do you have any clue why the callback is not executed?

On Android 12 anything works fine.

weliem, 2022-06-15:
I was checking the Android 13 developer pages but I don't see any changes related to Bluetooth. So it should work the same as on Android 12.

MDXDave, 2022-06-15:
Hm you write in a comment that some devices do not fire the ` onConnectionUpdated` callback. 
Maybe the Pixel 6 (w/ Android 13) is one of these devices?

weliem, 2022-06-17:
I update my Pixel 5 to Android 13 and did some testing. With my test app, the `onConnectionUpdated` is being called. It would surprise me if the Pixel 6 was behaving differently, but I guess it is still a possibility.

However, the right callback for knowing if you are connected or not, is not `onConnectionUpdated`. You should listen to `onConnectedPeripheral` in the `BluetoothCentralManagerCallback` interface. The `onConnectionUpdated` is for updates to the connection settings, like the timeout or phy.

weliem, 2022-07-02:
Any update on this issue?

closed 2022-07-21 by weliem

prepka-six15, 2022-08-31:
I'm having a similar issue. The `onPhyUpdate` and `onNotificationStateUpdate` callbacks aren't being made after calling `peripheral.setPreferredPhy(...)` or `peripheral.setNotify(...)`.

The `onMtuChanged` is being made after `peripheral.requestMtu(...)`.

In the log I see this interesting message:
bluetooth: packages/modules/Bluetooth/system/main/shim/acl.cc:758 OnPhyUpdate: Dropping ACL event with no callback
Which comes from here: https://android.googlesource.com/platform/packages/modules/Bluetooth/+/refs/heads/master/system/main/shim/acl.cc
I'm not really sure whats going on in this file...

I'm running on a Pixel 4XL with Android 13. I don't see this problem on a Pixel 4XL running Android 12.