weliem/blessed-android #149: The sample application does not allow connecting to a second device

rui-filipe-ribeiro, 2022-08-19:
I am using the sample application to study this excellent library.
It turns out that after connecting to one type of device, the application no longer works with another type, but it always works with the same device.
What do I need to do to allow connections with other devices without closing the application? Thanks.

weliem, 2022-08-19:
that's because the scanner is not re-started anymore. Just restart the scanner after the peripheral disconnected...

rui-filipe-ribeiro, 2022-08-20:
I tried this after device disconnected, even using a button to call explicitly, but it still doesn't recognize the second device. 

```
BluetoothCentralManager central = BluetoothHandler.getInstance(getApplicationContext()).central;
central.stopScan();
central.scanForPeripherals();
```

Any more ideas please?

rui-filipe-ribeiro, 2022-08-22:
After several attempts, I realized that I have to do the following:

To start scanning (example for glucose):

`
registerReceiver(glucoseDataReceiver, new IntentFilter(BluetoothHandler.MEASUREMENT_GLUCOSE));
BluetoothCentralManager central = BluetoothHandler.getInstance(getApplicationContext()).central;
central.scanForPeripheralsWithServices(new UUID[]{GLUCOSE_SERVICE_UUID});
`

to stop scanning:

`
unregisterReceiver(glucoseDataReceiver);
BluetoothCentralManager central = BluetoothHandler.getInstance(getApplicationContext()).central;
central.stopScan();
`

I hope this can help others.
Thanks for the initial tip,

closed 2022-08-22 by rui-filipe-ribeiro