weliem/blessed-android #150: Clarification of README on connectPeripheral

smashteevee, 2022-08-21:
Great write ups on Medium and thank you for this library! 

I wanted some clarification from two seeming contrary statements in your README:

First you say:

> - Supporting multiple simultaneous connections, so that you can connect to many peripherals

But later you say (emphasis mine):

> The method connectPeripheral will try to immediately connect to a device that has already been found using a scan. This method will time out after 30 seconds or less depending on the device manufacturer. Note that there can be **only 1 outstanding connectPeripheral**. So if it is called multiple times only 1 will succeed.

What do you mean you can only have 1 outstanding connectPeripheral? Do you mean that you cannot initiate a connectPeripheral command until any previous calls have returned either Connected/Failed/Disconnected?

Thanks!

smashteevee, 2022-08-22:
I read your Medium posts again and past issues. I think I understand the latter is correct because of the need to treat Gatt operations serially. But then what does your former statement refer to? That even though we should treat Gatt commands serially, the library supports interfacing multiple GATT connections simultaneously (it's just up to your own code to keep them in order)?

weliem, 2022-08-22:
Simply put:

- You can be **CONNECTED** to many devices at the time
- You can only be **CONNECTING** to 1 device at the time.....but only if you use connectPeripheral. This limitation is not there for autoConnectPeripheral

smashteevee, 2022-08-23:
Thanks for clarifying!

closed 2022-08-23 by smashteevee