weliem/blessed-android #151: startPairingPopupHack() doesn't seem to work anymore

SiNeumann, 2022-08-24:
I used a Pixel 5 to test regressions on Anroid 13. However I found that the "startPairingPopupHack()" doesn't seem to work anymore, instead Android will show 2 pairing notifications (1. confirm, 2. enter pin) which look exactly the same.
I checked the source at
https://android.googlesource.com/platform/packages/apps/Settings/+/master/src/com/android/settings/bluetooth/LocalBluetoothPreferences.java
And it seems that starting a discovery still should force a dialog

weliem, 2022-08-27:
The hack's intention is to make sure the popup are shown in the foreground rather than hidden in the notification bar. Seeing the popup twice is a different issue...

SiNeumann, 2022-09-06:
@weliem sorry for not describing it clearly, I got carried away with an implementation detail on my side.  
"Seeing the popup twice is a different issue" Yes I know, that is ok, I was saying twice as my testing device requires a pairing key, but it is not relevant to this issue.   
The issue is that the popup is not shown at all and instead a notification. For all my Android devices (Android  6-12) the PopUpHack (or using startDiscovery directly) always worked perfectly but does not for Pixel 5 with Android 13.  
The Pixel 5 is my canary device and as of now my only phone with Android 13. The popUpHack worked on this device previously.  
I'll look a bit more into this issue, collect logs and bugreports.

SiNeumann, 2022-09-06:
Getting closer now. While our ble-only peripheral advertises with the "General Discoverable" flag it is not found by "Discovery" in Android BluetoothSettings App (while it was in previous) versions, therefore (I assume)  it wasn't "discoverABLE" (whatever that means by Android Standards now) and therefore the PopUp won't show . I remember seeing this on iOS, which also ignores ble-only peripherals.
Anyway this has nothing to do with this library, therefore I close this issue.

Reopen, if I should keep you in the loop.

Thank you for your time.

closed 2022-09-06 by SiNeumann