weliem/blessed-android #153: onCentralConnected() is called when L2CAP CoC socket is connected

akwizgran, 2022-09-05:
Background: I'm working on a throwaway prototype to test BLE communication using L2CAP connection-oriented channels, falling back to GATT on devices that don't support CoC. The BLESSED library has saved me a lot of pain already - thank you!

I'm using BLESSED's CentralManager and PeripheralManager for scanning and advertising. The peripheral opens a CoC server socket and advertises the PSM in its scan response. The central uses the PSM to open a CoC socket to the peripheral via BluetoothDevice#createInsecureL2capChannel().

If device 1 advertises and device 2 scans, everything works as expected:
* Device 2 (operating as central) discovers device 1 (operating as peripheral)
* Device 2 opens a CoC socket to device 1
* Device 1's server socket accepts the incoming connection
* Device 1 gets a callback on BluetoothPeripheralManagerCallback#onCentralConnected()

However, when the socket is closed, there's no corresponding callback to onCentralDisconnected(). Does this mean there's actually still a connection between the devices, or is this a bug (either in Android's Bluetooth code or in BLESSED)?

If devices 1 and 2 both scan and advertise at the same time, and then one of them opens a CoC socket to the other, then *both* devices receive callbacks to onCentralConnected() when the CoC socket connects. Neither device receives a callback to BluetoothCentralManagerCallback#onConnectedPeripheral(). So it seems that both devices think they're connected to a central, but neither device thinks it's connected to a peripheral. When the socket is closed there's no corresponding callback to onCentralDisconnected() on either device. Does this mean there are actually still two connections between the devices (with each device operating in the central role for one connection and the peripheral role for the other)?

To reproduce the issue, I've created a branch of my prototype app with irrelevant code removed (min API level is 29 to ensure that connection-oriented channels are available):

https://code.briarproject.org/briar/public-mesh-testbed/-/tree/f88f9d5eb805ed34114b40e12b3149192a3e5b26

This is the class that does all the interaction with BLESSED:

https://code.briarproject.org/briar/public-mesh-testbed/-/blob/f88f9d5eb805ed34114b40e12b3149192a3e5b26/app/src/main/java/org/briarproject/publicmesh/bt/BtServiceImpl.java

To reproduce the first scenario above (one device advertising, one device scanning):
* Start the app on two devices (API >= 29) and grant permissions
* Tap "start advertising" on device 1, "start scanning" on device 2
* Wait for device 1 to appear in device 2's list of discovered devices (it should show the address, the PSM, and "Peripheral: Discovered")
* Optional: tap "stop scanning" on device 2
* In device 2's list of discovered devices, tap the entry corresponding to device 1
* The entry in device 2's list should show "L2CAP out: Connecting" and then "L2CAP out: Connected" (this reflects the outgoing CoC connection)
* An entry should appear in device 1's list with "L2CAP in: Connected" (this reflects the incoming CoC connection) and "Central: Connected" (because onCentralConnected() was called on device 1)
* The entry in device 2's list should still show "Peripheral: Discovered" (because onConnectedPeripheral() was *not* called on device 2)
* After 10 seconds the CoC socket will be closed
* The entry in device 2's list should show "L2CAP out: Disconnected" (because the CoC connection has been closed)
* The entry in device 1's list should show "L2CAP in: Disconnected" (because the CoC connection has been closed) and "Central: Connected" (because onCentralDisconnected() was *not* called on device 1)
* At this point, device 1 thinks it's connected to a central, but device 2 doesn't think it's connected to a peripheral

To reproduce the second scenario above (both devices advertising and scanning):
* Start the app on two devices (API >= 29) and grant permissions
* Tap "start advertising" on both devices
* Tap "start scanning" on both devices
* Wait for each device to appear in the other device's list of discovered devices (it should show the address, the PSM and "Peripheral: Discovered")
* Optional: tap "stop scanning" on both devices
* In device 2's list of discovered devices, tap the entry corresponding to device 1
* The entry in device 2's list should show "L2CAP out: Connecting" and then "L2CAP out: Connected" (this reflects the outgoing CoC connection)
* A second entry (with a different address) should appear in device 1's list (device 2 uses a different address for the outgoing CoC connection than the address it used for advertising, so device 1 sees two different addresses)
* The first entry in device 1's list should show "Peripheral: Discovered" (because onConnectedPeripheral() was *not* called on device 1)
* The second entry in device 1's list should show "L2CAP in: Connected" (this reflects the incoming CoC connection) and "Central: Connected" (because onCentralConnected() was called on device 1)
* The entry in device 2's list should show "Peripheral: Discovered" (because onConnectedPeripheral() was *not* called on device 2), and "Central: Connected" (because onCentralConnected() was called on device 2)
* After 10 seconds the CoC socket will be closed
* The entry in device 2's list should show "L2CAP out: Disconnected" (because the CoC connection has been closed) and "Central: Connected" (because onCentralDisconnected() was *not* called on device 2)
* The second entry in device 1's list should show "L2CAP in: Disconnected" (because the CoC connection has been closed) and "Central: Connected" (because onCentralDisconnected() was *not* called on device 1)
* At this point, both devices think they're connected to a central, but neither device thinks it's connected to a peripheral

weliem, 2022-09-07:
This library was designed for usage of GATT and not for CoC. I am not that experienced with CoC but from the documentation I understand that none of the GATT callbacks are relevant. So I am surprised to see that they sometimes get called by Android. I would advice: ignore them. For CoC you need to use the BluetoothSocket connection/close and anything else should be ignored....

akwizgran, 2022-09-07:
Thanks for the reply @weliem! Ignoring the callbacks sounds good, but since my app is using both GATT and CoC, I'm concerned that these callbacks may reflect changes to the internal state of BLESSED and/or the Android stack that could cause problems for the GATT side. For example, after disconnecting from a CoC peer the same device may want to connect to another peer via GATT. Still, I haven't seen any evidence of this yet, so I'll close this issue and open a new one if I find any problems of that kind.

closed 2022-09-07 by akwizgran