weliem/blessed-android #154: Library not loaded

watemotion, 2022-09-06:
Android Studio is not able to recognize the BluetoothHandler class.

My settings are the following:

**build.gradle project**

buildscript {
    repositories {
        google()
        mavenCentral()
    }

    dependencies {
        classpath "com.android.tools.build:gradle:7.0.4"

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

task clean(type: Delete) {
    delete rootProject.buildDir
}

**settings.gradle**

dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
    repositories {
        google()
        mavenCentral()
        jcenter() // Warning: this repository is going to shut down soon
        maven { url 'https://jitpack.io' }
    }
}

rootProject.name = "BlessedBLE"
include ':app'

and I've added the following dependencies in my build.gradle app

dependencies {

    implementation 'androidx.appcompat:appcompat:1.5.0'
    implementation 'com.google.android.material:material:1.6.1'
    implementation 'androidx.constraintlayout:constraintlayout:2.1.4'

    implementation 'com.github.weliem:blessed-android:2.3.0'
}

Anyway when I use

`
BluetoothHandler.getInstance(getApplicationContext());
`

the BluetoothHandler is not recognized as part of the Blessed library.

watemotion, 2022-09-06:
Oh I've found that the error has nothing to do with the library, you can discard my issue, thanks.

weliem, 2022-09-07:
Ok, closing then...

closed 2022-09-07 by weliem