weliem/blessed-android #158: Android 13 Callback Issues

prepka-six15, 2022-09-20:
I'm seeing the onPhyUpdate and onNotificationStateUpdate callbacks aren't being made after calling peripheral.setPreferredPhy(...) or peripheral.setNotify(...).

The onMtuChanged callback is being made after peripheral.requestMtu(...) however.

In the log I see this interesting message:
bluetooth: packages/modules/Bluetooth/system/main/shim/acl.cc:758 OnPhyUpdate: Dropping ACL event with no callback
Which comes from here: https://android.googlesource.com/platform/packages/modules/Bluetooth/+/refs/heads/master/system/main/shim/acl.cc
I'm not really sure whats going on in this file...

I'm running on a Pixel 4XL with Android 13. I don't see this problem on a Pixel 4XL running Android 12.

I'm using Blessed 2.3.1.

weliem, 2022-09-20:
I am aware of this. It has nothing to do with this library. I am running Android 13 on a Pixel 5 and I also don't get the callback. It's a bug in Andoird 13.... :-(

prepka-six15, 2022-09-20:
I see. Thanks for the response.

How should I verify that these changes have been made?
I suppose I could ask for the MTU last, and assume that once that finishes the others have either worked, or not.

prepka-six15, 2022-09-20:
I did some more tests. It's not that MTU works and the others don't.
It simply seems like requesting a PHY change breaks everything.
`peripheral.setPreferredPhy(PhyType.LE_2M, PhyType.LE_2M, PhyOptions.NO_PREFERRED);`
Anything I do before a PHY change works, anything after doesn't.

Doing the PHY change last also doesn't work. I still never see onPhyUpdate.
If I start talking to my device, I simply get writeCharacteristic timeouts.

Not changing the PHY works, but throughput seems to be MUCH slower as you would expect.

prepka-six15, 2022-09-20:
I see the library uses connectGatt with a transport value.
Maybe you also need a phy value now?
https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback)

weliem, 2022-09-21:
Just published a workaround in version 2.3.2

Can you give that a try?

prepka-six15, 2022-09-21:
The workaround works great!

Even though the callback doesn't happen my device is switching to `PhyType.LE_2M` and I'm getting the expected speedup.

For my own app, I added a simulated call to `onPhyUpdate(peripheral, PhyType.UNKNOWN_PHY_TYPE, PhyType.UNKNOWN_PHY_TYPE, GattStatus.SUCCESS)` when `(Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU)` so my app's logic works just like before.

DjNDB, 2024-03-02:
I just spent hours trying to get my app to work on Android 14, until I discovered that commenting out peripheral.setPreferredPhy() fixed it, which lead to me finding this issue on Android 13.

I fixed it by adding `Build.VERSION.SDK_INT == Build.VERSION_CODES.UPSIDE_DOWN_CAKE` to the timeout in `BluetoothPeripheral`.

I tested on a Pixel 4a and Samsung Galaxy Tab S5e, both on LineageOS 21, so I can't tell if Android 14 is affected as well in general, or if it's a LineageOS issue.

```java        
public boolean setPreferredPhy(@NotNull final PhyType txPhy, @NotNull final PhyType rxPhy, @NotNull final PhyOptions phyOptions) {
	Objects.requireNonNull(txPhy);
	Objects.requireNonNull(rxPhy);
	Objects.requireNonNull(phyOptions);

	if (Build.VERSION.SDK_INT < Build.VERSION_CODES.O) {
		Logger.e(TAG, "setPreferredPhy requires Android 8.0 or newer");
		return false;
	}

	return enqueue(new Runnable() {
		@Override
		public void run() {
			if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
				currentCommand = SET_PHY_TYPE_COMMAND;
				Logger.i(TAG, "setting preferred Phy: tx = %s, rx = %s, options = %s", txPhy, rxPhy, phyOptions);
				bluetoothGatt.setPreferredPhy(txPhy.mask, rxPhy.mask, phyOptions.value);

				if (Build.VERSION.SDK_INT == Build.VERSION_CODES.TIRAMISU || Build.VERSION.SDK_INT == Build.VERSION_CODES.UPSIDE_DOWN_CAKE) {
					// There is a bug in Android 13 where onPhyUpdate is not always called
					// Therefore complete this command after a delay in order not to block the queueu
					currentCommand = IDLE;
					callbackHandler.postDelayed(new Runnable() {
						@Override
						public void run() {
							completedCommand();
						}
					}, 200);
				}
			}
		}
	});
}
```

weliem, 2024-03-03:
thx for the info, will look into it

closed 2024-07-06 by weliem