weliem/blessed-android #162: Possibility of exposing BluetoothDevice

sylwia-golonko, 2022-10-14:
`BluetoothCentral` currently exposed only mac addres, name and mtu. Can the library also expose the whole `BluetoothDevice`? It contains a lot of useful methods (e.g. for bonding) that can't be accessed via `BluetoothCentral` .

weliem, 2022-10-14:
I added a method getBondState() to BluetoothCentral and released version 2.3.3.

I don't think there is much more interesting stuff you can ask BluetoothDevice. What else do you need?

sylwia-golonko, 2022-10-17:
Things related to bonding, like:
- force bonding (`createBond`), at least there is another way to force it on server side 
- accept pairing popup `setPairingConfirmation` (system app)

Thanks in advance.

weliem, 2022-10-31:
just released a new version that also offers createBond() and setPairingConfirmation()....2.3.5

weliem, 2022-11-03:
closing this issue since requested functions are now available...

closed 2022-11-03 by weliem