weliem/blessed-android #163: Devices Nearby Permission in Android 12

Leopold84, 2022-10-26:
Hi Weliem,
I have the following problem: I have an Android app with Target SDK 29 that I can't switch to a higher Target SDK at the moment for a number of reasons. This app uses GPS and Bluetooth Classic and therefore requires ACCESS_FINE_LOCATION permissions. Now to my actual problem that I observed with Android 12: As soon as I integrated the Blessed Library (v2.0.11), the Devices-Nearby Permission is not granted by default on Android 12 Devices. This Permission must be granted manually, otherwise the app crashes when the App tries to establish  a BT connection. Also, due to Target SDK 10, I can't programmatically check this permission. Do you  have an idea how I can solve this?

weliem, 2022-10-27:
Can you try to upgrade to the latest Blessed version? 2.0.11 does not support Android 12...

Leopold84, 2022-10-29:
OK, thank you.

closed 2022-10-29 by Leopold84