weliem/blessed-android #168: Notify of Characteristic Change Fails on 2.4.0

sean7512, 2022-11-17:
When using version 2.3.4, everything works fine; however updating to 2.4.0 breaks notifying.

Inside of ```BluetoothPeripheralManager``` is the following:

```java
public boolean notifyCharacteristicChanged(@NotNull final byte[] value, @NotNull final BluetoothGattCharacteristic characteristic) {
    Objects.requireNonNull(value, CHARACTERISTIC_VALUE_IS_NULL);
    Objects.requireNonNull(characteristic, CHARACTERISTIC_IS_NULL);

    if (doesNotSupportNotifying(characteristic)) return false;

    boolean result = true;
    for (BluetoothCentral device : getConnectedCentrals()) {
        if (!notifyCharacteristicChanged(value, device, characteristic)) {
            result = false;
        }
    }
    return result;
}
```

When using 2.4.0, the ```getConnectedCentrals()``` call is returning an empty list; however in 2.3.4, it is returning the centrals correctly.

sean7512, 2022-11-17:
I think the call in the loop may need changed to ```getCentralsWantingNotifications()```. Unfortunately, I have run out of time today to continue testing.

weliem, 2022-11-26:
I think this is related to the other issues you are reporting. In 2.4.0 the logic for getting connected devices was changed and now uses the library internal lists rather than the OS.

sean7512, 2022-11-29:
I don't think so, because it doesn't work even if I turn off advertising. There are 2 methods for notifying of characteristic changed, 1 that takes a central to notify and 1 that doesn't. On the one that doesn't, it is still querying the OS and not using the internal list, as it should. When using the debugger, ```getCentralsWantingNotifications(characteristic)``` returns the expected values, but the method is calling ```getConnectedCentrals()``` which is returning an empty list.

weliem, 2022-11-29:
I tried the example ([https://github.com/weliem/bluetooth-server-example](https://github.com/weliem/bluetooth-server-example)) and everything seems fine, but that is of course only a peripheral.

sean7512, 2022-12-01:
Another interesting finding.....so it gets removed from the internal list because of a random onConnectionStateChange callback that says it was disconnected. If you use ```bluetoothManager.getConnectionState(device, BluetoothProfile.GATT)``` in that call, it will return that it is connected, contradicting the notification. I wonder if we can only fall down and call ```handleDeviceDisconnected(device)``` if the OS tells us it was disconnected?

archie94, 2022-12-14:
Is there any update on this? We are facing this problem too.

archie94, 2022-12-14:
I had to revert to version `2.1.2` 
We were not getting this problem on debug variants but on release variants (`2.4.0` and `2.3.5`) `onCharacteristicUpdate` was not being called.

archie94, 2022-12-16:
Umm .. this one

```
    public void onCharacteristicUpdate(@NotNull BluetoothPeripheral peripheral, @NotNull byte[] value, @NotNull BluetoothGattCharacteristic characteristic, @NotNull GattStatus status) {}
```

in `BluetoothPeripheralCallback`

Also may be related but from Nordic BLE library:

https://github.com/NordicSemiconductor/Android-BLE-Library/pull/395

weliem, 2023-01-04:
@archie94 , you are referring to an issue with a BluetoothPeripheral but @sean7512 is referring to an issue when notifying to a BluetoothCentral....

Please open a new issue if you still are having this issue.

strike76, 2023-01-14:
> I had to revert to version `2.1.2` We were not getting this problem on debug variants but on release variants (`2.4.0` and `2.3.5`) `onCharacteristicUpdate` was not being called.

Same problem here. No problem with debug release, i need to use 2.2.3 version

sean7512, 2024-12-02:
I just wanted to leave this update for anyone that is still affected by this. The new Kotlin version (https://github.com/weliem/blessed-kotlin) does NOT have the issue described here and this can likely be closed sinceI believe the Java version is essentially EOL'd.