weliem/blessed-android #172: onCharacteristicUpdate not calling.

karthick1611, 2022-12-28:
I'm sending some commands and write was success but not able to read the data. Here I attach my logs.

`BluetoothPeripheral     com.library.blenew                   I  connected to 'GNRGYBT_HMCNF2000000' (NONE) in 0.2s
2022-12-28 17:28:40.922  3525-3525  BluetoothPeripheral     com.library.blenew                   I  discovering services of 'GNRGYBT_HMCNF2000000' with delay of 0 ms
2022-12-28 17:28:40.922  3525-3525  BluetoothGatt           com.library.blenew                   D  discoverServices() - device: 50:02:91:81:21:E6
2022-12-28 17:28:41.375  3525-3539  BluetoothGatt           com.library.blenew                   D  onConnectionUpdated() - Device=50:02:91:81:21:E6 interval=6 latency=0 timeout=500 status=0
2022-12-28 17:28:41.407  3525-3539  BluetoothPeripheral     com.library.blenew                   I  connection parameters: interval=7.5ms latency=0 timeout=5s
2022-12-28 17:28:41.612  3525-3539  BluetoothGatt           com.library.blenew                   D  onSearchComplete() = Device=50:02:91:81:21:E6 Status=0
2022-12-28 17:28:41.619  3525-3539  BluetoothPeripheral     com.library.blenew                   I  discovered 3 services for 'GNRGYBT_HMCNF2000000'
2022-12-28 17:28:41.623  3525-3525  Found                   com.library.blenew                   E  Bluetooth4
2022-12-28 17:28:41.623  3525-3525  Read                    com.library.blenew                   E  6e400003-b5a3-f393-e0a9-e50e24dcca9e
2022-12-28 17:28:41.624  3525-3525  Write                   com.library.blenew                   E  6e400002-b5a3-f393-e0a9-e50e24dcca9e
2022-12-28 17:28:41.625  3525-3525  BluetoothGatt           com.library.blenew                   D  configureMTU() - device: 50:02:91:81:21:E6 mtu: 517
2022-12-28 17:28:41.630  3525-3525  BluetoothPeripheral     com.library.blenew                   I  requesting MTU of 517
2022-12-28 17:28:41.652  3525-3539  BluetoothGatt           com.library.blenew                   D  onConfigureMTU() - Device=50:02:91:81:21:E6 mtu=517 status=0
2022-12-28 17:28:41.654  3525-3525  BluetoothGatt           com.library.blenew                   D  requestConnectionPriority() - params: 1
2022-12-28 17:28:41.658  3525-3525  BluetoothPeripheral     com.library.blenew                   I  requesting connection priority HIGH
2022-12-28 17:28:41.691  3525-3539  BluetoothGatt           com.library.blenew                   D  onConnectionUpdated() - Device=50:02:91:81:21:E6 interval=36 latency=0 timeout=500 status=0
2022-12-28 17:28:41.695  3525-3539  BluetoothPeripheral     com.library.blenew                   I  connection parameters: interval=45.0ms latency=0 timeout=5s
2022-12-28 17:28:41.803  3525-3536  System                  com.library.blenew                   W  A resource failed to call close. 
2022-12-28 17:28:42.161  3525-3525  BluetoothGatt           com.library.blenew                   D  setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2022-12-28 17:28:42.186  3525-3539  BluetoothGatt           com.library.blenew                   D  onConnectionUpdated() - Device=50:02:91:81:21:E6 interval=12 latency=0 timeout=500 status=0
2022-12-28 17:28:42.188  3525-3539  BluetoothPeripheral     com.library.blenew                   I  connection parameters: interval=15.0ms latency=0 timeout=5s
2022-12-28 17:28:42.205  3525-3525  TAG                     com.library.blenew                   E  SUCCESS: Notify set to 'on' for 6e400003-b5a3-f393-e0a9-e50e24dcca9e
2022-12-28 17:28:42.228  3525-3525  BluetoothPeripheral     com.library.blenew                   I  writing <7375204d616b654c6f74734f664073> to characteristic <6e400002-b5a3-f393-e0a9-e50e24dcca9e>
2022-12-28 17:28:42.255  3525-3525  Write Success           com.library.blenew                   I  SUCCESS: Writing <7375204d616b654c6f74734f664073> to <6e400002-b5a3-f393-e0a9-e50e24dcca9e>
2022-12-28 17:33:41.021  3525-3525  ViewRootIm...nActivity] com.library.blenew                   I  handleAppVisibility mAppVisible=true visible=false
2022-12-28 17:33:41.047  3525-3547  OpenGLRenderer          com.library.blenew                   D  setSurface called with nullptr
2022-12-28 17:33:41.048  3525-3547  OpenGLRenderer          com.library.blenew                   D  setSurface() destroyed EGLSurface
2022-12-28 17:33:41.048  3525-3547  OpenGLRenderer          com.library.blenew                   D  destroyEglSurface`

weliem, 2022-12-28:
I don't see anything strange in the logs. Are you sure the write you are doing will lead to a notification?

karthick1611, 2022-12-29:
`                val currentTimeCharacteristic = peripheral.getCharacteristic(BLUETOOTH_LE_NRF_SERVICE, readCharacteristic?.uuid!!)
                if (currentTimeCharacteristic != null)
                    peripheral.setNotify(currentTimeCharacteristic, true)
`

`            override fun onNotificationStateUpdate(
                peripheral: BluetoothPeripheral,
                characteristic: BluetoothGattCharacteristic,
                status: GattStatus
            ) {
                if (status == GattStatus.SUCCESS) {
                    if(peripheral.isNotifying(characteristic)) {
                        Log.e("TAG", String.format("SUCCESS: Notify set to 'on' for %s", characteristic.uuid))
                        writeCharacteristic?.let { sendStringToBle(peripheral, it) }
                    } else {
                        Log.e("TAG", String.format("SUCCESS: Notify set to 'off' for %s", characteristic.uuid))
                    }
                } else {
                    Log.e("TAG", String.format("ERROR: Changing notification state failed for %s", characteristic.uuid))
                }
            }
`

`            override fun onCharacteristicWrite(
                peripheral: BluetoothPeripheral,
                value: ByteArray,
                characteristic: BluetoothGattCharacteristic,
                status: GattStatus
            ) {
                if (status == GattStatus.SUCCESS) {
                    Log.i("Write Success", String.format("SUCCESS: Writing <%s> to <%s>", asHexString(value), characteristic.uuid))
                } else {
                    Log.i("Write Failed", String.format("ERROR: Failed writing <%s> to <%s> (%s)", asHexString(value), characteristic.uuid, status))
                }
            }
`

karthick1611, 2022-12-29:
How can I make sure that will lead to a notification?.

karthick1611, 2022-12-29:
override fun onCharacteristicUpdate(
                peripheral: BluetoothPeripheral,
                value: ByteArray,
                characteristic: BluetoothGattCharacteristic,
                status: GattStatus
            ) {
                if (status != GattStatus.SUCCESS) return
                val characteristicUUID = characteristic.uuid
                val parser = BluetoothBytesParser(value)
                Log.e("Response", value.decodeToString())
                if (characteristic == readCharacteristic) {
                    Log.e("Response", value.decodeToString())
                }
            }

weliem, 2022-12-31:
Whether or not a notification will happen depends on your device and the protocol it uses. You need to know the protocol in order to properly coomunicate with the device.

closed 2023-01-03 by weliem

karthick1611, 2023-01-04:
Thanks for the response. I figured out the issue and it's resolved. Thanks again.