weliem/blessed-android #173: If you use CentralManager and PeripheralManager together, BluetoothCentralManager.cancelConnection() does not work.

bagmin33, 2023-01-02:
I am not good at English. Sorry

If you use BluetoothCentralManager and BluetoothPeripheralManager together, 
The BluetoothCentralManager.cancelConnection() function does not work.

`
    public void initBluetooth(Context context) {
        bluetoothManager = (BluetoothManager) context.getSystemService(Context.BLUETOOTH_SERVICE);
        bluetoothAdapter = bluetoothManager.getAdapter();
        bluetoothCentralManager = new BluetoothCentralManager(context, bluetoothCentralManagerCallback, new Handler(Looper.getMainLooper()));
        bluetoothPeripheralManager = new BluetoothPeripheralManager(context, bluetoothManager, bluetoothPeripheralManagerCallback);
    }
`

1. call bluetoothCentralManager.cancelConnection() 
2. response BluetoothCentralManagerCallback.onDisconnectedPeripheral() status is SUCCESS 
3. but It was not actually disconnected. (T^T)

Please help me.

weliem, 2023-01-02:
Can you provide a log so that I can see what is going on?

weliem, 2023-03-26:
Closing due to lack of information

closed 2023-03-26 by weliem