weliem/blessed-android #175: In Samsung S22 series smartphones not working.

karthick1611, 2023-01-04:
Hi,

I tried implement your library, everything working fine but In Samsung S22 series smartphones (Android 13), after sometime it's disconnected.

For example, I writing some commands to my charger and I'm able to read the response as well. That's not a issue but after some seconds, writing getting failed and bluetooth getting disconnected.

But In other Android 13 devices and Samsung devices, it's working fine.

Is that any issue in Samsung?.

karthick1611, 2023-01-04:
Same commands what I'm sending working perfectly in Other devices. And it's good to know, u r developing for S22. May be, u'll get to know about this issue. While sending the data to my charger, while writing the data Bluetooth disconnected and writing failed.

weliem, 2023-01-04:
well, there is one problem with the S22 and that is changing the MTU. It will say that it works, but it doesnt. You can check by removing the requestMtu() call so that it stays on the default 23.

karthick1611, 2023-01-04:
Well.. Well.. Well.. It's working now but there's a issue as well. 

value byte array is longer than allowed by MTU, write will fail if peripheral does not support long writes

Getting this and not getting response fully.

karthick1611, 2023-01-11:
I didn't mention there is issue with this library. 

And Thanks for the replies bro.

karthick1611, 2023-01-12:
As far as I tested, I believe Request MTU not working in Bluetooth Version 5.2. Is anyone faced the same issue?.

weliem, 2023-03-26:
Closing this issue since it relates to S22 bug and is not a library issue

closed 2023-03-26 by weliem