weliem/blessed-android #177: onCharacteristicUpdate() Callback is not being called for specific length of data received.

traxformania, 2023-02-03:
Hello, I'm having a strange problem that I can't solve and I can't understand the reason, and I need your support.

First of all I'm testing your library with Android 8 and Android 13 devices and all of them have this problem. 

I am trying to communicate a phone (Which uses your library) with a device of our own manufacture via BLE. 

After I connect my phone to the other device, I'm getting write & read characteristics and enabling read characteristic's notify property to receive data over onCharacteristicUpdate callback metod.

I'm setting MTU size to 250, and requestConnectionPriority to ConnectionPriority.BALANCED.

I'm sending a command ranging from 1 to ~320 in length from the phone to the other device, and expect a response that also varies in length from 1 to ~320. (in multiples of 16)

Other device always receives my commands without any problem. But phone not receiving whole response from other device sometimes.

In my test, If other device tries to send 

284 byte => onCharacteristicUpdate callback is called 2 times and i get the whole response.
268 byte => onCharacteristicUpdate callback is called 2 times and i get the whole response.
252 byte => onCharacteristicUpdate callback is called 1 time and i don't get the whole response. Second part is missing.
236 byte => onCharacteristicUpdate callback is called 2 times and i get the whole response.
...
...
Everything works fine below (236 bytes). (The response length is reduced by 16 bytes each.)

The strange thing is that this problem always occurs while waiting for 252 bytes on the same phone. 
If I change requestConnectionPriority to ConnectionPriority. HIGH it works sometimes.
If I change MTU size to other values, this problem occurs at different response lengths.
And this problem also varies from phone to phone.

If I use an app called "Serial Bluetooth Terminal" from play store to test I don't have this problem. This app always receives full response so I think problem's source is not other device.

What can I try for solution?

weliem, 2023-02-04:
Can you explain why your device is sending 2 notifications after you do a write? Is the device sending you the same as what you wrote?

Anyway, your problem is related to your MTU. You say you choose 250 as the MTU and then you get problems with sending 252. So that is the first one after your max MTU. I guess if you lower the MTU you see the problem in the next size larger than your MTU?

When you try to write a byte array larger than your MTU, a 'long write' is done. Basically, under the hood, Android does multiple writes and your device re-assembles it to the original byte array. So it looks like the long writes are functioning well. It is just that at 252 bytes it should already do a long write and perhaps Android doesn't do that. The Blessed library doesn't do anything special when doing a write....it just calls the standard Android writeCharacteristic method.

The only thing I don't get is why it would work with 'Serial Bluetooth Terminal' on the same phone. Can you reconfirm that?

Also, if you can, can you share some logs?

weliem, 2023-03-21:
Closing issue due to lack of information.....

closed 2023-03-21 by weliem