weliem/blessed-android #179: Bluetooth Connection Issues

thy-hoang-nn, 2023-03-09:
Hello, I am interested in your library and have used it to create my application.

Is there anything to check when Bluetooth is turned off, though? I'm referring to when your Bluetooth LE device is moved far from your phone and there is a connectivity problem. If you maintain your devices within Bluetooth range, they will be linked and their data, including any previously unsent data, will be sent to your phone.

I hope you will reply as soon as possible and thanks for other supports.

weliem, 2023-03-09:
You can use central.isBluetoothEnabled()

weliem, 2023-03-21:
When a device gets out of range, the connection is automatically broken. Similarly when BLE is turned off, the connection will be lost...

closed 2023-03-21 by weliem