weliem/blessed-android #180: Failed to write when enabling listening

TheLittleZhe, 2023-03-21:
Hello, I have read your blog, deeply admire your explanation, and have tried to introduce this library. However, now I am constantly switching on/off Bluetooth, and often fail to write 2902 attributes when notifying, and then Bluetooth is disconnected. What may be the reason for this? My model is Huawei P40

TheLittleZhe, 2023-03-21:
`    override fun onConnectedPeripheral(peripheral: BluetoothPeripheral) {
        super.onConnectedPeripheral(peripheral)
        LogUtil.log("connected")
        peripheral.setNotify(serviceBondUUID, characteristicBondUUID,true)
        Thread.sleep(200)
        peripheral.setNotify(serviceUUID, characteristicUUID,true)
        Thread.sleep(200)
        peripheral.requestMtu(160)
    }` 
this is my code, it'll notify and set mtu when connected

TheLittleZhe, 2023-03-21:
this is my code ,it'll notify and set mtu when connected
`
    override fun onConnectedPeripheral(peripheral: BluetoothPeripheral) {
        super.onConnectedPeripheral(peripheral)
        LogUtil.log("connected")
        peripheral.setNotify(serviceBondUUID, characteristicBondUUID,true)
        Thread.sleep(200)
        peripheral.setNotify(serviceUUID, characteristicUUID,true)
        Thread.sleep(200)
        peripheral.requestMtu(160)
    }
`

weliem, 2023-03-21:
You have several issues:
- move you code to onServicesDiscovered.
- remove the sleep() statements as each command will be queued anyway

TheLittleZhe, 2023-03-21:
Thank you very much for the reminder, I have modified the code and it works fine, but now I have a new problem. I read the source code and found that as long as the binding status is "BONDING", some operations will be ignored, and this seems to be a problem on my test machine, because its binding status keeps returning to "BINDING", even if I have set it on the page I canceled the pairing, so I commented out the judgment of the binding state in nextCommand, but every time I connect to BLE, it will actively disconnect. What might be the cause of this?

TheLittleZhe, 2023-03-21:
For business process reasons, my project needs to notify two features before pairing. These two features will send me the encrypted pairing code and the random number for calculating the pairing code, so I cannot notify after pairing operate

weliem, 2023-03-21:
I don't think your idea will work. It is not my library that handles the actual bonding. Android does it itself. So if bonding is started it must complete before anything else can happens You cannot first enable some notifications and use those values in the bonding process if your peripheral already demanded that bonding is necessary.

The only thing you can do it to make sure your peripheral allows a connection without bonding. Then turn on the notifications and get the pairing code. And then start the bonding by calling createBond()....

Or...I am not understanding the situation correctly. If so please show a complete log so I can see what is happening.

TheLittleZhe, 2023-03-22:
Thank you very much for your help, I think I have solved my problem, the key reason is the bluetooth hardware, I have no problem at all with another car-grade bluetooth peripheral, I try to get the hardware developer to help me debug, It was found that the Bluetooth connection was actively disconnected, and everything is in good condition.

closed 2023-03-22 by TheLittleZhe