weliem/blessed-android #186: some phone cant open bonding window

EdgeYu, 2023-06-07:
Hello: 
     I really have a very strange problem, the specific manifestation is that the phone does not pop up the pairing box, I think it is not a blessed problem, because I tested thirty phones, but only on one phone this problem appeared, strangely, on this phone another bluetooth test tool app can trigger pairing, I tried call createBond()  to actively trigger pairing, but the effect is not good (sometimes the delay is high), I would like to ask if there is any relevant experience in this situation that can be used for reference or advice?

EdgeYu, 2023-06-07:
i checked the HIC and stack log, it's not into Security verification link（only in my app）

weliem, 2023-06-08:
I have never seen the issue you describe on my test phones. Are you sure the popup is not accessible via the swipe down menu.

EdgeYu, 2023-06-09:
thanks for your reply,I figured out the phenomenon, but I can't explain why.
you can read this code:
`  if (Build.MODEL == "PEQM00" && Build.BRAND == "OPPO") {
                peripheral.requestNotify(serviceUUID, characteristicUUID, true)
            } else {
                peripheral.setNotify(serviceUUID, characteristicUUID, true)
            }`
in OPPO Reno6,have this question, i use blessed setNotify to open the BLE notify,but its not work,so i try to use GATT directly:
`    private fun BluetoothPeripheral.requestNotify(
        uuid: UUID,
        characteristicUUID: UUID,
        enable: Boolean
    ) {
       
        this.bluetoothGatt?.let { gatt ->
            val service = gatt.getService(uuid)
            if(gatt.setCharacteristicNotification(service.getCharacteristic(characteristicUUID), enable)){
                val defaultUUID = UUID.fromString("00002902-0000-1000-8000-00805f9b34fb")
                val descriptor = service.getCharacteristic(characteristicUUID).getDescriptor(defaultUUID)
                descriptor.value = if (enable) BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE else BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE
                gatt.writeDescriptor(descriptor)
            }
        }
    }`
it's work.

EdgeYu, 2023-06-09:
my BLE chip is KW36, It gives developers control over when pairing is triggered.
So I set the pairing to trigger when I receive a Notify request.
i checked  HCI log, the notify request success, and my chip received notify packet, im not sure this question from where, if this is hardware question, why its work when i change the notify way

weliem, 2023-12-11:
this really sounds like a phone-specific problem.... I am afraid I cant help

closed 2023-12-11 by weliem

TheLittleZhe, 2024-06-17:
@weliem hey weliem，I am the one who raised this question. My friend’s account was used above. I am currently replying to you in person.

TheLittleZhe, 2024-06-17:
hope this will help more people