weliem/blessed-android #190: ESP32 BLE connected but SERVICE_UUID and notification not received

sdranju, 2023-10-09:
I'm having a tough time for Android BLE with ESP32 BLE. 
I can successfully connect ESP32 but I'm not getting SERVICE_UUID from the device. I can see from the serial monitor that ESP32 is connected and sends notification with the interval. However, I'm not able to receive it in Android.
I added my ESP32 code sample here, hope someone can help.

```
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
BLECharacteristic* pCharacteristic_2 = NULL;
BLEDescriptor *pDescr;
BLE2902 *pBLE2902;

bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

#define SERVICE_UUID        "0000181D-0000-1000-8000-00805f9b34fb"
#define CHAR1_UUID          "00002A9D-0000-1000-8000-00805f9b34fb"
#define CHAR2_UUID          "00002A9D-0001-1000-8000-00805f9b34fb"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class CharacteristicCallBack: public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pChar) override { 
    std::string pChar2_value_stdstr = pChar->getValue();
    String pChar2_value_string = String(pChar2_value_stdstr.c_str());
    int pChar2_value_int = pChar2_value_string.toInt();
    Serial.println("pChar2: " + String(pChar2_value_int)); 
  }
};

void setup() {
  randomSeed(analogRead(A0)); // Use an analog reading as the seed value

  Serial.begin(115200);
  delay(10);
  Serial.println("\n");

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  pCharacteristic = pService->createCharacteristic(
                      CHAR1_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_NOTIFY
                    );                   

  pCharacteristic_2 = pService->createCharacteristic(
                      CHAR2_UUID,
                      BLECharacteristic::PROPERTY_WRITE  
                    );  

  // Create a BLE Descriptor
  pDescr = new BLEDescriptor((uint16_t)0x2901);
  pDescr->setValue("A very interesting variable");
  pCharacteristic->addDescriptor(pDescr);
  
  pBLE2902 = new BLE2902();
  pBLE2902->setNotifications(true);
  
  // Add all Descriptors here
  pCharacteristic->addDescriptor(pBLE2902);
  pCharacteristic_2->addDescriptor(new BLE2902());
  
  // After defining the desriptors, set the callback functions
  pCharacteristic_2->setCallbacks(new CharacteristicCallBack());
  
  // Start the service
  pService->start();

  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
  BLEDevice::startAdvertising();
  Serial.println("Waiting a client connection to notify...\n");

}

void loop() {
    // notify changed value
    if (deviceConnected) {
        value = random(100, 10000);  // Simulated weight data
        String weightStr = String(value);  // Convert weight to a string

        pCharacteristic->setValue(value);
        //pCharacteristic->setValue(weightStr.c_str());
        pCharacteristic->notify();
        Serial.println("Weight Notified: " + weightStr + " g");
        delay(1000);
    }

    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("Start advertising...\n");
        oldDeviceConnected = deviceConnected;
    }

    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }

}

```

weliem, 2023-10-12:
This repository is about an Android library. Can't help you with your ESP code....

What does you Android code look like?

sdranju, 2023-10-15:
@weliem 
Thank you very for your feedback. I got it fixed. Your library only accepts data in byte array from the BLE device. But was sending it either string or int format.
```
        pCharacteristic->setValue(value);
        //pCharacteristic->setValue(weightStr.c_str());
```

closed 2023-10-21 by weliem